openbmc/openbmc #267: Implement IPMI Network Settings Override

anoo1, 2016-05-09:
From Jeremy regarding the IPMI boot options:

we may want to implement the "network settings
override" functionality in SL GA2 here. That is an OEM boot parameter
selector, using parameter ID 0x61.

The first byte of the parameter data is the same as parameter ID 5 (ie,
the valid flag in the top bit). Persistency doesn't matter here, as
petitboot will not clear out this setting.

The rest of the parameter data is a host network settings description.
For other BMC firmware, the BMC just treats this as opaque (the BMC will
just return the same data (including that data1 byte), in response to a
Get System Boot Option). For OpenBMC, we may want the BMC to know how to
construct that data; the spec for that is at:

http://ozlabs.au.ibm.com/~sam/ipmi-network-format

shenki, 2016-05-10:
The URL is not accessible

@sammj can you publish the content?

sammj, 2016-05-11:
Ah, that link was more an internal draft, but re-reading it it looks sane to me - I've copied it over to http://ozlabs.org/~sam/ipmi-network-format
It's really aimed more at the user-side, the BMC probably only really cares about the format of the "set" request header (0x00 0x08 0x61 0x80) and "get" request header (0x00 0x09 0x61 0x80 0x00), the rest is up to Petitboot to interpret.

Petitboot support was first added here: https://github.com/open-power/petitboot/commit/1e1e3417262e44531d2a5c6dd3152d5deb92fb90
The request format is also described here more informally: https://sthbrx.github.io/blog/2016/03/21/no-network-for-you/

sammj, 2016-05-11:
>  Persistency doesn't matter here, as petitboot will not clear out this setting.

Also persistence _does_ matter, as Petitboot respects it as of https://github.com/open-power/petitboot/commit/79f738d13fa491abac82c77ac42f59aef4217b1d , but I believe it reads the persistent bit from the 'data1' byte, not the first byte. Still, it should probably be set to non-persistent in the first byte for sanity :)

williamspatrick, 2016-06-15:
@ratagupt - You have a commit for this already?  Please add to your commit message something like:

'Resolves openbmc/openbmc#267'

closed 2016-09-02 by williamspatrick