openbmc/openbmc #268: Add a cap to number of event logs

anoo1, 2016-05-10:
In issue https://github.com/openbmc/openbmc/issues/257, a system logged over 1200 event logs, which are too many to be useful for debug.
The event log manager caps the number of logs based on filesystem space, but the hardware logs are small and a lot of them can fit in the allocated event log space.
This issue is to add a cap on the number of event logs in addition to the existing space cap.

Kenthliu, 2016-05-11:
Does this commit make the cap?
https://github.com/openbmc/openbmc/commit/6a861ddd11cca8558f371ade53ab43c7855f5a6d

anoo1, 2016-05-11:
That commit adds a cap to the total size the event logs take up, regardless of how many. If there are hostboot eSEL events which are bigger, there would only be a few saved. But in this case the HW events are smaller, so a lot more can fit in the allocated space. So we need to add a cap to the number of logs as well even if the allocated space has not been filled.

causten, 2016-05-11:
Limiting the number of logs requires some additional thoughts.  What do you want to drop?  Oldest log, new log coming in, severity level based ?

Kenthliu, 2016-05-12:
Traditonally, BMC drop the oldest log when SEL is full.

causten, 2016-05-17:
I'm putting in a fix.  Like the size limiter the change will stop recording logs.  It will proclaim to the syslog that it did not record the log.  
https://github.com/openbmc/phosphor-event/pull/18

williamspatrick, 2016-05-26:
Are we set to close this?

anoo1, 2016-05-26:
Available in openbmc tag 0.8.

closed 2016-05-26 by anoo1