openbmc/openbmc #272: Define autorestart retry limit and recovery for phosphor processes

anoo1, 2016-05-16:
Autorestart was added with https://github.com/openbmc/openbmc/pull/149 .
Next step is to define a retry limit for each process so that if there's an issue, the faulty process stops trying to restart after a number of retries. Also for some processes, after or instead a retry, a different recovery may be desired like rebooting the BMC or logging an event.

geissonator, 2017-08-28:
# **STATUS**
- 8/28: Focused on previous sprint stories and build stability, have not started

geissonator, 2017-09-26:
Discussed on mailing list here: https://lists.ozlabs.org/pipermail/openbmc/2017-September/009087.html
Put code in for it here: https://gerrit.openbmc-project.xyz/#/c/6540/

We are going to hold off on doing any automatic reboot actions for specific services until we get more learning with openbmc.  We can open up a follow up issue if we decide we'd like to put in service-specific actions.

closed 2017-09-26 by geissonator