openbmc/openbmc #273: Implement user roles

anoo1, 2016-05-16:
Implement user roles (via whitelist first, then based on the Redfish role model), to allow the ability to make the DBUS interfaces private.

williamspatrick, 2016-05-26:
@anoo1 - Any assessment on if this will be done for 1.0?

agangidi53, 2016-06-01:
Some of our internal developers feel that we should manage BMC users via ipmitool commands from Petitboot / Host etc., 
I guess there's security concerns doing it that way though. 
What's your initial train of thought in terms of how to manage users ?

williamspatrick, 2016-06-03:
We were planning on using PAM users and groups on the BMC to define and configure users and roles.  We had also planned on making a dbus / REST API for managing them.  Eventually we may migrate to the Redfish-way when we support Redfish.

Doing it from the host via ipmitool would require us to define and implement new OEM commands.  We had not planned to do that.  The in-band path is unauthenticated as well, so pretty much anyone who has host access could define new users and assign them superuser roles on the BMC.  I know we will have the in-band restricted mode filtering, but it still seems dangerous to expose up to the host.

williamspatrick, 2016-06-15:
This should be broken down into a few stories:

- PAM manipulation.
- REST server support.  (IPMI, Redfish server supports?)
- Dbus APIs for manipulating PAM groups.
- Dbus API annotations.

williamspatrick, 2016-06-30:
Also need to understand the IPMI requirements.  They have some plaintext hashing algorithm that is used to validate users.  We want to have a common user repository.

sivassrr, 2016-07-24:
Brief Description of the issue: 
With the help of Rest and Dbus-send test able to set value for sensors related to CPU DIMM core present and fault. Also able to set value for OperatingSystem status sensor, OCC enable sensor, Boot Progress and Boot Count sensors. 

Expected Behavior:
Using REST, Dbus-send command only getting the value of sensor interfaces are fine. But not setting the same.

Actual Behavior:
Able to set all sensors value using both Dbus-send and REST method. 

Recreations Steps:
Run the test https://github.com/sivassrr/SivasPublic/blob/master/test.sensors.robot
https://github.com/sivassrr/SivasPublic/blob/master/test_sensors_dbus.robot
Tested on Barreleye/Palmetto and VERSION_ID="v1.0-94-g612a690-dirty".

geissonator, 2017-05-24:
Also be sure to track removal of the old org.openbmc.UserManager here.

dkodihal, 2017-11-16:
1) User management API
a) Define API to represent a User.
b) Define API to represent a Group.
c) API to add/modify/remove Users (this need not be methods, could just be a "Create" with a)
d) API to add/modify/remove Groups ((this need not be methods, could just be a "Create" with b).
e) Should an API user be query/control to all that he can do via linux user-management tools, by logging on the BMC .
f) Define API to assign user roles (could just be a property in a) ).
g) API to provide LDAP server config.
h) API to manipulate PAM?

2) User management via LDAP
a) Implement LDAP server config API.
b) Login code to talk to LDAP server.
c) User management API code from 1) to talk to LDAP server.
d) Get LDAP libs onto the image.

3) PAM based authentication
a) Deploy PAM config files.
b) Implement PAM manipulation APIs.
c) What does systemd have to offer in terms of PAM? Do we need to implement our own plugin?
d) Integrate pam_ldap
e) /login implementaton to use PAM.

rfrandse, 2018-08-17:
moved to https://github.com/ibm-openbmc/dev/issues/83

closed 2018-08-17 by rfrandse