openbmc/openbmc #278: Add checkstop sensor to non-Barreleye platforms

anoo1, 2016-05-19:
Change https://github.com/openbmc/skeleton/pull/74 added the handling of the checkstop sensor so that the BMC resets the system if the OS is hung. It also added the checkstop sensor to Barreleye.
Need to look up the specifications for the other supported openbmc platforms like palmetto, garrison, etc and add the checkstop sensor as well so that they make use of this function for handling OS checkstops.

anoo1, 2016-07-06:
Added with https://github.com/openbmc/skeleton/commit/b4140b888f586c98c5d03893166c5074a98cf951

closed 2016-07-06 by anoo1

williamspatrick, 2016-07-06:
Oops.  We should have put that in the commit message.

'Closes openbmc/openbmc#278.'

anoo1, 2016-07-06:
Yeah, next time 👍