openbmc/openbmc #280: PreTag 0.8 Testing failure for 'Set the power with string of characters'

causten, 2016-05-19:
Robot Test Case failed
Full Name: 	Tests.Test Generic Conf.Set the power with string of characters
Documentation: 	

**BAD PATH** This test case set the power values with string of characters Expectation is to return error.
Start / End / Elapsed: 	20160519 15:17:43.333 / 20160519 15:17:46.618 / 00:00:03.285
Status: 	FAIL (critical)
Message: 	''abcdefg'=='abcdefg'' should not be true.

rahulmah, 2016-06-14:
@ratagupt : Can you please look at this failure. I think this test case was never working.

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #17](https://github.com/openbmc/openbmc-test-automation/issues/17) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick

gkeishin, 2016-08-19:
https://github.com/openbmc/openbmc-test-automation/issues/17#issuecomment-240934248