openbmc/openbmc #281: PreTag 0.8 Testing failure for 'Set the power with greater then MAX_POWER_VALUE'

causten, 2016-05-19:
Full Name: 	Tests.Test Generic Conf.Set the power with greater then MAX_POWER_VALUE
Documentation: 	

**BAD PATH** This test case sets the power value which is greater then MAX_ALLOWED_VALUE,Expectation is to return error
Start / End / Elapsed: 	20160519 15:17:46.619 / 20160519 15:17:50.565 / 00:00:03.946
Status: 	FAIL (critical)
Message: 	1010 == 1010

rahulmah, 2016-06-14:
@ratagupt : It seem like we are able to set power with value greater then 1000(like 1010 of your test case). Please let us know if a test case problem or it is fw issue.

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #18](https://github.com/openbmc/openbmc-test-automation/issues/18) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick