openbmc/openbmc #285: PreTag 0.8 Testing failure for 'Blink Slow the Identify LED'

causten, 2016-05-19:
Blink Slow the Identify LED
Full Name: 	Tests.Test Led.Blink Slow the Identify LED
Documentation: 	

This testcase is to test the setBlinkSlow functionality of the Identify LED. The LED state is sampled to figure out whether the LED is blinking. There is no distinguishing between whether the LED is blinking fast or slow for this testcase to pass.
Start / End / Elapsed: 	20160519 15:21:21.448 / 20160519 15:21:42.413 / 00:00:20.965
Status: 	FAIL (critical)
Message: 	'0 > 0 and 5 > 0' should be true.

vishwabmc, 2016-05-24:
Alright... I feel its not a bug wrt the LED blinking.. It should be about how we are sampling it. Doing it via network REST way definitely adds onto time. I would visually verify if the LED is blinking slow and if there are issues we can look into it. The duty cycle ratio was agreed with Norm after he visually inspected the LEDs.. 

Or are you saying its just about the Automated Testability request  ?

causten, 2016-05-24:
Can the user use rest to discover if the led is set to what they asked it to be set to?  The test case indicates they can not.  So how does the user determine if the Led is blinking via rest ?
 
 Sent from my iPhone using IBM Verse

vishwabmc, 2016-05-24:
So the user can use REST to see if its blinking looking at the pattern.. REST way can not today tell if its binking fast or slow since there would always be these network odds.. So back then, the requirement was to see if its blinking.. If we need at the granularity of duty cycle, we need to put some change into code and return what is the state  by looking at our own settings since those settings are what is directing the LEDs. 

We can take this change .. but would be a change to suffice testability aspects only.

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #22](https://github.com/openbmc/openbmc-test-automation/issues/22) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick