openbmc/openbmc #286: PreTag 0.8 Testing failure for 'Set IP address on valid Interface'

causten, 2016-05-19:
Full Name: 	Tests.Test Networkd.Set IP address on valid Interface
Documentation: 	
This test case sets the ip on the interface and validates that ip address has been set or not. Expectation is the ip address shoTags: 	GOOD-PATH
Start / End / Elapsed: 	20160519 15:22:29.625 / 20160519 15:22:29.627 / 00:00:00.002
Status: 	FAIL (critical)
Message: 	Environment variable 'NEW_BMC_IP' does not exist.uld get added.

anoo1, 2016-05-23:
@rahulmah seems the env variable doesn't exist in the test case? Could you take a look?

rahulmah, 2016-05-24:
This seem to have happen because NEW_BMC_IP env variable was not initilize (using export) before running this test case.

rahulmah, 2016-05-24:
@anoo1 This seem to have happen because NEW_BMC_IP env variable was not initilize (using export) before running this test case.

mdmillerii, 2016-05-25:
The testcase should check its requirements and print an clear message when they are not met.

Does the test suite need a configuration file or wrapper script to prompt for this information?

causten, 2016-05-25:
tox.ini is a wrapper file that holds system type specific information.  The I can foresee a need for a system specific file.  It would be useful for other issues too like ensuring that 3 different IO or Memory configs all are validated correctly.

I'd like it if we put some thought in to how to do this correctly that supports configs like manual, automation runs and other companies wanting to run the suite

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #16](https://github.com/openbmc/openbmc-test-automation/issues/16) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick