openbmc/openbmc #287: PreTag 0.8 Testing failure for 'get names /'

causten, 2016-05-19:
Full Name: 	Tests.Test Obmcrest.get names /
Start / End / Elapsed: 	20160519 15:23:39.010 / 20160519 15:23:53.120 / 00:00:14.110
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='9.3.23.24', port=443): Max retries exceeded with url: /enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.24', port=443): Read timed out. (read timeout=5.0)",))
00:00:14.109 KEYWORD ${resp} = rest_client . OpenBMC Get Request /enumerate

anoo1, 2016-05-23:
What is the test case trying to do? @rahulmah could you take a  look?

rahulmah, 2016-05-24:
@anoo1 : 'get names /' internally fo GET for /enumerate

}rahuls-MacBook-Pro:~ rahulmaheshwari$ curl -c cjar -b cjar -k -X GET -d "{\"data\":[]}" https://9.3.23.30/enumerate
{
  "data": {
    "/org/openbmc": {}, 
    "/org/openbmc/HostIpmi/1": {}, 
    "/org/openbmc/NetworkManager/Interface": {}, 
    "/org/openbmc/UserManager/Group": {}, 
    "/org/openbmc/UserManager/Groups": {}, 
    "/org/openbmc/UserManager/User": {}, 
    "/org/openbmc/UserManager/Users": {}, 
    "/org/openbmc/buttons": {}, 
    "/org/openbmc/buttons/power0": {
      "state": 0, 
      "timer": 0
    }, 
    "/org/openbmc/buttons/reset0": {
      "state": 0,

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #23](https://github.com/openbmc/openbmc-test-automation/issues/23) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick