openbmc/openbmc #289: PreTag 0.8 Testing failure for 'Set System Time after setting epoch time'

causten, 2016-05-19:
Set System Time after setting epoch time
Full Name: 	Tests.Test Time.Set System Time after setting epoch time
Documentation: 	

**GOOD PATH** This test case tries to set system time to epoch time using ipmi and then tries to set to to valid time. Expectation is that new time can be set on a system with epoch time.
Start / End / Elapsed: 	20160519 15:28:14.471 / 20160519 15:28:15.108 / 00:00:00.637
Status: 	FAIL (critical)
Message: 	Set SEL Time command failed: Unspecified error: 1 != 0

anoo1, 2016-05-23:
If I remember correctly, there were some issues setting the epoch time so we were not going to have test cases for it. @tomjoseph83 could you take a look?

rahulmah, 2016-05-24:
@anoo1 : I have open below pull request to remove this epoch test case. 
https://github.com/mkumatag/openbmc-automation/pull/54

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #24](https://github.com/openbmc/openbmc-test-automation/issues/24) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick