openbmc/openbmc #290: PreTag 0.8 Testing failure for 'user/group authentication'

causten, 2016-05-19:
6 failures, I am using one issue to resolve all

Create and delete user with user group name
Full Name: 	Tests.Test User.Create and delete user with user group name
Documentation: 	

**GOOD PATH** This testcase is for testing user creation with user name, password, comment and group name(user group) in open bmc.
Start / End / Elapsed: 	20160519 15:28:31.740 / 20160519 15:28:40.484 / 00:00:08.744
Status: 	FAIL (critical)
Message: 	Authentication failed for user 'HuHkThIS'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Create multiple users
Full Name: 	Tests.Test User.Create multiple users
Documentation: 	

**GOOD PATH** This testcase is to verify that multiple users creation in open bmc.
Start / End / Elapsed: 	20160519 15:28:40.484 / 20160519 15:28:48.145 / 00:00:07.661
Status: 	FAIL (critical)
Message: 	Authentication failed for user 'mrnviL4h'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Create and delete user without password
Full Name: 	Tests.Test User.Create and delete user without password
Documentation: 	

**GOOD PATH** This testcase is to create and delete a user without password in open bmc.
Start / End / Elapsed: 	20160519 15:28:48.146 / 20160519 15:28:57.151 / 00:00:09.005
Status: 	FAIL (critical)
Message: 	Authentication failed for user 'nxeznfvp'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 Set password for existing user
Full Name: 	Tests.Test User.Set password for existing user
Documentation: 	

**GOOD PATH** This testcase is for testing password set for user in open bmc.
Start / End / Elapsed: 	20160519 15:28:57.152 / 20160519 15:29:04.658 / 00:00:07.506
Status: 	FAIL (critical)
Message: 	Authentication failed for user 'BaToCj2H'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Set password with empty password for existing
Full Name: 	Tests.Test User.Set password with empty password for existing
Documentation: 	

**GOOD PATH** This testcase is to verify that empty password can be set for a existing user.
Start / End / Elapsed: 	20160519 15:29:04.658 / 20160519 15:29:12.031 / 00:00:07.373
Status: 	FAIL (critical)
Message: 	Authentication failed for user '6mm7Lxrk'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

rahulmah, 2016-06-14:
@hramasub : Please provide update on this issue.

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #14](https://github.com/openbmc/openbmc-test-automation/issues/14) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick