openbmc/openbmc #292: Fan Control fails to set Fan speed to safe setting during Reboot / Firmware Update

agangidi53, 2016-05-20:
**Current**: Fan Control fails to set Fan speed to safe setting before going into a Reboot / Firmware Update.  This is important because we leave the system at risk if the Host load ramps up while BMC is rebooting (~2 minutes) or during a firmware update via /run/initramfs/image-******  (~14 minutes).

**Expectation**: Fan control sets the Fans to max speed before going down and takes back the control after rebooting and ramps it down to appropriate speeds.

anoo1, 2016-05-23:
hi @Kenthliu , is this something your team can help with?

johnhcwang, 2016-05-25:
Is it possible that we can call fan_control.py to set full speed in systemd-reboot.service?

Kenthliu, 2016-05-27:
Hi @anoo1, Ok we will handle this feature.
Hi @agangidi53 , the timing of fan speed you concern is during bmc reboot, right? Now we are investigating how to make fan full speed after reboot. All the update methods need to reboot BMC finally. Reboot BMC can use ipmi command in host or RestAPI, those will call warmreset dbus method. We propose to make fan full speed after warmreset was called. So the correct update procedure will be reboot BMC through ipmi command or RestAPI in the final step. Reboot directly under BMC debug console will not work. Is this OK with you? Thanks.

agangidi53, 2016-06-06:
I am sorry for delayed reply. I was out of office for sometime and only checked this now. 
I don't understand setting fan speed to max shouldn't work via reboot via console. 
What is the difference between rebooting via RestAPI and rebooting via the console that makes fan speeds to be set to max easier than the other  ?
What is needed for you to get this to work on both ways ?

williamspatrick, 2016-06-30:
Is this fixed in the Foxconn tree?

Kenthliu, 2016-07-01:
Yes

msbarth, 2018-01-24:
With the ability to configure a watchdog and/or set speed events within a yaml file for fan control, the conditions noted here can be resolved to set fan speed to a given target under these conditions. i.e.) For the witherspoon system, a watchdog is triggered when the BMC is rebooted which drives fans to 100%PWM until the fan control application returns to active state.
In another case, a property can be configured in the fan control yaml to trigger setting the fans to a specific speed, for example..in prep of a code update to occur.

rfrandse, 2018-01-24:
<https://gerrit.openbmc-project.xyz/8626> Support properties of type string
Resolves  #292 Fan Control fails to set Fan speed to safe setting during Reboot / Firmware Update

closed 2018-01-29 by geissonator

rfrandse, 2018-03-21:
<https://gerrit.openbmc-project.xyz/9650> Changing the text Quiesced to Error
Fixes  #292 Fan Control fails to set Fan speed to safe setting during Reboot / Firmware Update