openbmc/openbmc #293: Flashing BMC via /run/initramfs : No way to monitor if firmware update fails

agangidi53, 2016-05-20:
For the BMC firmware update: If we copy images to /run/initramfs (via REST API Flashing method OR Manual copy) and reboot there is no way to monitor how the flash method is progressing or if it has failed.

This becomes super important specially since it takes about 12-15 minutes for the BMC to come back up on the network if we choose this flash method and hence poses a huge risk in a Data Center environment of going into a state where inserting a debug cable to press "Ctrl D" or to monitor exact error becomes necessary.

We should design a way to switch into some what of a 'rescue shell' before we flash the individual BMC images. This environment should let us monitor the flash update procedure via Network.

mdmillerii, 2016-05-23:
Lets discuss your desired update sequence.

The update script can be run anytime the corresponding part of the flash is not being used by the kernel for read-on-demand paging.  The `openbmc-init-fs` `init` script already has configuration hooks to copy the read-only file system image to ram, or to download it from a network via `tftp` or `wget` via `http` or `ftp` (`https` requires `openssl` helper in the read-only filesystem).   The `init` script also has hooks to copy files from the `whitelist` from the read-write overlay filesystem to ram and back at shutdown.  The `update` script was originally written as an extension to the `shutdown` script and it needs a significant error handling design change.  The `init` script can be run from the `shutdown` state if the same kernel is being used, but would require flags to say don't really `kexec` or similar to get though and unpatched `systemd`.  The `update` script today will run if both the readonly and readwrite filesystems are the ram variants.

1. Is a reboot to prepare for update after setting a firmware environment variable acceptable?

2. Should we shutdown and restart on the current kernel?  Should it be with the current file system or the new one?  Might there be compatibility issues with a new fs and old kernel?

3. Would you prefer to netboot the kernel and new filesystem?  Run them from RAM and write the flash in the background?  This would get the system running on the new image as fast as possible.

4. Would you prefer to always copy files to ram to allow a new image be written at any time?

5.  Does the `update` script need to recognize which partitions are mounted (in-use) so we can continue to run from the full read-write overlay (which preserves command history etc for the root user, and also would for example preserved the erasure of logs by the rotation vs simple copy back of files) while the other flash portions are updated?   This requires parsing `sysfs` for the partition relationships (the `bmc` is a superset of the others) but that information is now available in the 4.4 kernel.

agangidi53, 2016-06-14:
Milton

In general, is there a way we can have basic network services and ssh service enabled completely out of RAM ( not use flash so that there's no read-on-demand paging) and then use something like pflash for update.  

**Is a reboot to prepare for update after setting a firmware environment variable acceptable?**
Does this mean we reboot BMC and boot into some "safe mode shell" where we can flash the BMC and monitor it and then boot from newly flashed BMC image?
Generally this is acceptable. All we'd like is to know the status of BMC update after reboot command is pressed and BMC update is happening. This may NOT necessarily mean that we need a console access while update is happening but we'd like to know at least the following 
a) If update is in progress or hanged because of a error ?
b) If it is in progress what image (full bmc image or kernel, u-boot etc)  is it updating ?
c) A way to query the key events / percentage progress of update. 

**Should we shutdown and restart on the current kernel? Should it be with the current file system or the new one? Might there be compatibility issues with a new fs and old kernel?**
We don't have strict preferences on this. As long as we have some way to know firmware update progress. But I understand when you talk about compatibility issues. Booting with new kernel and New FS might be better  

**Would you prefer to netboot the kernel and new filesystem? Run them from RAM and write the flash in the background? This would get the system running on the new image as fast as possible.**
This sounds like a interesting idea but this might need a DHCP / TFTP server running in Data Center on the out of band network.  That is something that's generally discouraged. Is having a DHCP / DFTP serv

**Would you prefer to always copy files to ram to allow a new image be written at any time?**
This is not a strict requirement not preference towards this or probably even against it.

**Does the update script need to recognize which partitions are mounted (in-use) so we can continue to run from the full read-write overlay (which preserves command history etc for the root user, and also would for example preserved the erasure of logs by the rotation vs simple copy back of files) while the other flash portions are updated? This requires parsing sysfs for the partition relationships (the bmc is a superset of the others) but that information is now available in the 4.4 kernel.**

Preserving command history and logs sounds like a good idea. How much of a pain point is this to implement ?

mdmillerii, 2016-06-15:
I have implemented planned enhancements to the update script so it can be used while the system is running the normal stack but in a dedicated mode in #297.  The script also can be used to validate a set of update images.  

The 
 thought from here would be to rename the current flash bmc method to be unpack, then add validate and apply methods, along with a abort or cancel that would remove pending updates.  A prepare would round out the list and can set the u-boot variables and reboot or perhaps just stop and restart user space by defining kexec with some hooks.

Returning status can be a capture of the update script output or could be filtered to remove past progress reports.

I haven't started these updates yet and am willing to collaborate (take help) on the implementation.  Or you can tell me you are happy with coping the images and running the update script over ssh.

Regarding ssh at update environment, busy box has a telnetd and a basic passwd file is in the initramfs but I haven't yet packaged sshd and it's dependencies.  A file list could be generated to copy from the running system to /run/initramfs but I haven't recognized the worth.  HTTP FTP and TFTP are available but HTTPS requires openssl.

Regarding shell history, the file is not in the base white list but I plan to rebase #180 when I can update bmc control to make the same change, then you can add a file with any additional files and directories you want saved over a rwfs layer purge.

I'll probably edit this when I have access to a full browser and keyboard.

agangidi53, 2016-06-20:
@mdmillerii So this dedicated mode you talk about is the same one that helps a manufacturers flash the BMC via host ? Where is the reference code (which repo) for base implementation of this dedicated mode ? I'd like to know the services that run in said mode.

mdmillerii, 2016-06-20:
@agangidi53 The services that run are exactly the same as a normal boot, the difference is the file systems are not mounted from the flash by the `init` script based on options passed to it via the `u-boot-env`.  Instead, a copy of the `rofs` partition is copied into memory and mounted via the `loopback` driver with the `mount -o loop` option, and the whitelisted files from the `rwfs` partition are copied into ram and then written back to the file system on an orderly shutdown.  (I did not implement trimming the `image-rofs` to the size of the `squashfs` file system image but keep the full partition).

This update mode is set via the code added in openbmc/openpower-host-ipmi-oem#9 and cleared by the `clear-once` service added in 32906e3037190dfb3fdae1787b812ea5da35d0a6 via pull #200.  

With my changes in #297, this __could__ also be controlled by updating the the `initramfs` partition with an image that contains these variables, and then later applying one without, I plan to implement the rest interface `prepare` method using the same `openbmconce` variable in `u-boot`.

My attempts at writing documentation for the `init` and `update` scripts and the boot process keep getting delayed fixing bugs.  I don't have my current rough draft with me at the current time.

closed 2016-07-08 by jk-ozlabs