openbmc/openbmc #295: Flashing takes a really long time

shenki, 2016-05-21:
Flashing the BMC using our update method takes far longer than using pflash. This ticket is to investigate why and fix it if possible.

Doing this will mitigate issues such as https://github.com/openbmc/openbmc/issues/293

causten, 2016-05-21:
There certainly is a chip technology component to this problem, there are parts that know how to fast erase

mdmillerii, 2016-05-23:
There are several of known reaasons:

1. The kernel only handles one erase block size (plus whole flash erase).  To support 4k erase sectors the kernel is configured for 4k blocks.  `pflash` opportunistically uses larger blocks.
2.  There is a large variation in flash chip technology: the `MX25L25635E` from `MACRONIX INTERNATIONAL CO., LTD.` has 4k sector erase at 60/300ms typical/max (thats .06/.3s) and 64k at .7/2s, while the `N25Q256A` from `Micron Technology` has similar 64k sector erase of .7/3s but the sub-sector 4k is .25/.8s.  Thus the penalty for item 1 abvoe is more significant.
3. The flash_cp command uses a separate system call for each erase block, so even if we looked for larger block opportunistically, we would not find them.  There is commented code in `busybox` `flash_cp` command to do larger blocks but the comment lists reasons for being rejected.
4. There is or was a problem with multiple sector erase in one system call that has not been diagnosed.  That is why `fw_setenv` is configured to use a shadow ram file that is copied back on shutdown with `flash_cp`.  Before I put in the latest memcpy_fromio workaround I was finding the flash to raise flash programing sequence errors.  And yes I should open an issue ;-).

williamspatrick, 2016-05-26:
I could see this being deferred to 2.0, but leaving in 1.0 for now.

geissonator, 2017-10-13:
The UBI based flash is quite quick now but we still have a lot of systems using the older update method.

amboar, 2017-10-16:
Specifically, all systems that aren't Witherspoon.

rfrandse, 2018-08-20:
moved to https://github.com/ibm-openbmc/dev/issues/132

closed 2018-08-20 by rfrandse