openbmc/openbmc #298: add diagnostic and development tools

legoater, 2016-05-24:
Here is an initial list : 

- gard from https://github.com/open-power/skiboot
- pdbg from https://github.com/open-power/pdbg

shenki, 2016-05-25:
Untested bitbake files are in this tree: https://github.com/shenki/openbmc/tree/openpower-tools

If someone can show me how to build the one package with bitbake that would be great.

williamspatrick, 2016-05-26:
Do we want this in the flash all the time or do we need to create a tools tarball?  I'm concerned about using the flash space for a development-only tool.

shenki, 2016-05-26:
The problem with making them not-default is they are never there when you need them.

gard makes sense to be on there all the time I think.

pdbg is a bit harder to sell. But it's 18K :)

legoater, 2016-05-26:
This is part of a BMC survival kit. I think strace would be nice for user space also.

williamspatrick, 2016-06-14:
Need to define what we are going to put into the flash image for debug tools and what mechanism we use to add an extra set of debug tools for developers to side-load.

williamspatrick, 2016-08-04:
I am adding strace to the SDK as a tool.  When we figure out the mechanism to side-load applications (ex. tarball for NFS?) this should be moved into that set.

https://gerrit.openbmc-project.xyz/#/c/126/

williamspatrick, 2016-08-05:
That previous approach of adding strace to the SDK did not work.  So, I figured out how to make tarballs instead.

https://gerrit.openbmc-project.xyz/140

williamspatrick, 2016-08-06:
The following commits add support asked for here:
    * https://gerrit.openbmc-project.xyz/140 meta-phosphor: Create debug tarball image
    * https://gerrit.openbmc-project.xyz/141 obmc-phosphor-image: Enable debug tarball
    * https://gerrit.openbmc-project.xyz/142 meta-openpower: Add pflash to debug tarball
    * https://gerrit.openbmc-project.xyz/145 pflash: Refactor recipe
    * https://gerrit.openbmc-project.xyz/146 skiboot-gard: Add recipe and add to tarball
    * https://gerrit.openbmc-project.xyz/147 meta-openpower: Add pdbg tool recipe
    * https://gerrit.openbmc-project.xyz/148 meta-openpower: Add pdbg to tarball

This creates a debug tarball with strace globally and {pflash, gard, pdbg} on Power machines.  I included pflash here even though it is already part of the flash image because we could potentially remove pflash in the future as we refactor openbmc/skeleton content.

williamspatrick, 2016-08-11:
Code has been merged now.

closed 2016-08-11 by williamspatrick