openbmc/openbmc #299: MBE error events DIMM number is wrong

Kenthliu, 2016-05-24:
Run test OpenPOWER RAS Verification Guide Version 1.0 -- section 3.4 Trial 4: Memory DIMM UE Checkstop Error.
Use "putscom -c 0x80000001 0201145E 0x0840000000000000" to trigger MBE. on memory buff ID 80000001.
RestApi will record two MBE error events on the same DIMM. There are no DIMM 1,4,5 events.
And check the RestApi DIMM iverntory information. DIMM 0,1,4,5 has fault present.
Blow in openbmc event
[image]
[image]
[image]

causten, 2016-05-24:
send me the logs, I'll figure it out
How are you injecting the errors?

Kenthliu, 2016-05-25:
Please reference section 3.4 Trial 4 in below file.
[RASverif_22January2016_pub.pdf](https://github.com/openbmc/openbmc/files/280815/RASverif_22January2016_pub.pdf)

causten, 2016-05-25:
Ken can you add the log to box for me to view?
 
 
Chris AustenPOWER Systems Enablement Manager(512) 286-5184 (T/L: 363-5184)

Kenthliu, 2016-05-25:
Yes @causten Here you are. event data with journctl log.

[dimm.txt](https://github.com/openbmc/openbmc/files/280962/dimm.txt)
[image]

causten, 2016-06-16:
I have had some long discussions with the hostboot team about this issue.  https://github.com/open-power/hostboot

I key off of the data they send me.  The big event log that has all the debug data useful to the hardware engineers to debug.  The eSEL which is sent to me does not contain multiple FRU callouts.  There is no plan to do it either.  

Unfortunately this means it is 'working as designed'

Kenthliu, 2016-06-17:
It sounds like a limitation. Did you open a issue on https://github.com/open-power/hostboot? Can you please provide more detail information about why no other DIMM# log? Then we can close this issue.

anoo1, 2016-06-20:
Opened hostboot issue: https://github.com/open-power/hostboot/issues/60

Kenthliu, 2016-06-22:
Thanks for the explain.
Also record Patirck's response here.

A memory UE is expected to not be isolated to a single DIMM, so there is always multiple callouts. If I remember right, Hostboot does two things when there is an error with a guard event. 

Send down a SEL/eSEL. 
Assert fault sensors for the associated FRUs.
Since a SEL is associated with a sensor, in effect, the SEL also points to a FRU. What Chris was observing is only one FRU from the SEL path but all the FRUs from the fault sensor path. Since we treat sensors as "inventory" and SELS as "error events", which are handled by two different pieces of code, and the fault sensor doesn't actually have any correlation back to the SEL anyhow, the BMC is unable to show an association between the SEL and all the callout-FRUs; we only get the one for the SEL. 
This information is contained deep in the eSEL, which the BMC is not presently able to parse and would require a bunch of HUID->FRU-ID mapping on BMC side anyhow. As best as we can tell there is no way to get this information in a pure IPMI sense.

closed 2016-06-22 by Kenthliu