openbmc/openbmc #300: Segfault during reboot

KennethWilke, 2016-05-24:
On a server using the v0.6.1 BMC version, we encountered a segfault when rebooting the BMC. I did happen to have a debug cable plugged into this server so I have this log of the occurrence:

After hitting this it seems the file system it tries to mount from /dev/mtdblock6 is corrupt and the BMC hit's the emergency maintenance shell. I can drop a debug cable in my own boxes to fix this, but need to make sure this doesn't occur on servers in a remote datacenter.

```
root@barreleye:~# reboot
[  OK  ] Stopped target Multi-User System.
         Stopping DBUS introspecting REST server....
         Stopping Phosphor Inverse ARP daemon...
         Stopping Kernel Logging Service...
[  OK  ] Stopped target Timers.
         Stopping Phosphor OpenBMC user management daemon...
         Stopping Network DBUS object...
         Stopping Phosphor OpenBMC DBus REST daemon...
         Stopping Phosphor OpenBMC event management daemon...
[  OK  ] Stopped target Network.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped target System Time Synchronized.
         Stopping Settings DBUS Object...
[  OK  ] Stopped target Remote File Systems.
         Stopping Login Service...
         Stopping Phosphor OpenBMC BT to DBUS...
         Stopping System Logging Service...
[  OK  ] Removed slice system-getty.slice.
         Stopping Network Service...
         Stopping Phosphor OpenBMC IPMI daemon...
[  OK  ] Stopped target Login Prompts.
         Stopping Serial Getty on ttyS4...
[  OK  ] Stopped Settings DBUS Object.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped DBUS introspecting REST server..
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Network DBUS object.
[  OK  ] Stopped Phosphor OpenBMC user management daemon.
[  OK  ] Stopped Phosphor Inverse ARP daemon.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Phosphor OpenBMC event management daemon.
[  OK  ] Stopped Phosphor OpenBMC BT to DBUS.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped Phosphor OpenBMC IPMI daemon.
[  OK  ] Stopped Phosphor OpenBMC DBus REST daemon.
         Stopping Phosphor OpenBMC DBus service management daemon...
[  OK  ] Stopped Clear one time boot overrides.
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping D-Bus System Message Bus...
         Stopping Temp placeholder for skeleton function...
[  OK  ] Stopped Phosphor OpenBMC DBus service management daemon.
[  OK  ] Stopped Temp placeholder for skeleton function.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Closed Syslog Socket.
[  OK  ] Closed dropbear.socket.
[  OK  ] Stopped target System Initialization.
         Stopping Update UTMP about System Boot/Shutdown...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Network Time Synchronization...
         Stopping Load/Save Random Seed...
[  OK  ] Stopped target Swap.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Dispatch Password Requests to Console Directory Watch.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Stopped Create Volatile Files and Directories.
[... 94 lines cut ...]
[<c005ecfc>] (delete_from_page_cache) from [<c0069430>] (truncate_inode_page+0x98/0xa4)
[<c0069430>] (truncate_inode_page) from [<c00695d4>] (truncate_inode_pages_range+0x168/0x518)
[<c00695d4>] (truncate_inode_pages_range) from [<c0069998>] (truncate_inode_pages+0x14/0x1c)
[<c0069998>] (truncate_inode_pages) from [<c00a1060>] (evict+0x90/0x128)
[<c00a1060>] (evict) from [<c00a1128>] (dispose_list+0x30/0x3c)
[<c00a1128>] (dispose_list) from [<c00a1d64>] (evict_inodes+0xb4/0xbc)
[<c00a1d64>] (evict_inodes) from [<c008e5d0>] (generic_shutdown_super+0x40/0xc0)
[<c008e5d0>] (generic_shutdown_super) from [<c008e830>] (kill_block_super+0x18/0x68)
[<c008e830>] (kill_block_super) from [<c008e930>] (deactivate_locked_super+0x44/0x74)
[<c008e930>] (deactivate_locked_super) from [<c00a42d4>] (cleanup_mnt+0x4c/0x70)
[<c00a42d4>] (cleanup_mnt) from [<c002a97c>] (task_work_run+0x6c/0x80)
[<c002a97c>] (task_work_run) from [<c000cba8>] (do_work_pending+0xa4/0xc0)
[<c000cba8>] (do_work_pending) from [<c000a2d0>] (slow_work_pending+0xc/0x20)
Code: e121f002 e594300c e3530000 ba000000 (e7f001f2)
---[ end trace b045e78c59981ae5 ]---
Segmentation fault
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=126368k,nr_inodes=31592,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
```

shenki, 2016-05-25:
@mdmillerii is this related to the corruption caused by i2c?

mdmillerii, 2016-05-26:
Its hard to say definitively but I hope so.

The first error detected was a bad page while tearing down `systemd-journal`:

BUG: Bad page map in process systemd-journal  pte:4d69a100 pmd:4dddd831
addr:7f600000 vm_flags:00000875 anon_vma:  (null) mapping:cf4e5974 index:3e
file:systemd-journald fault:filemap_fault mmap:generic_file_readonly_mmap readpage:squashfs_readpage
CPU: 0 PID: 531 Comm: systemd-journal Not tainted 4.3.6-openbmc-20160222-1 #1

The vma say sits a file map page from squashfs in a read-only mmap.

The PTE is only PTE_USER not PRESENT, YOUNG, DIRTY, RDONLY, XN, or SHARED.

The next prints about rss counters say we had 1 file page present and -1 swap pages present, because we didn't classify that as a pte.

The BUG at shutdown says we still have a rmap entry for a PTE.

The address is nicely aligned, so all this goes along with the first byte of a page being zeroed, which may be the 32k i2c rom being read.

Looks like @jk-ozlabs has a patch coming soon.

mdmillerii, 2016-06-01:
This is the same issue as #190, just a different file.  It's still likely corruption of the page table.

williamspatrick, 2016-06-14:
Is this resolved?

mdmillerii, 2016-06-14:
This should be fixed by the i2c fix in openbmc-20160606-1 merged in the main branch via #313.
I believe that tag should have alternative patches to support occ with 12-core parts and the scu changes.  The JFFS2 patches are in the kernel v4.4.y stream.
Alternatives to the fixes to u-boot in 0.6.1 were merged in #304.

closed 2016-06-15 by williamspatrick