openbmc/openbmc #306: Create error log on single FRU resulted failed due to Read time out error

sivassrr, 2016-05-26:
BMC Name:  9.3.23.32

Ran entire suite today and seeing read time out error in the test create error log on single FRU. 
===================Details as follows===========================================
TEST Create error log on single FRU
Full Name: 	Tests.Test Association.Create error log on single FRU
Documentation: 	

**GOOD PATH** Create an error log on single FRU and verify its association.
Start / End / Elapsed: 	20160526 09:13:47.442 / 20160526 09:14:05.839 / 00:00:18.397
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /org/openbmc/records/events/action/clear (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",))
00:00:18.396 KEYWORD Clear all logs
Start / End / Elapsed: 	20160526 09:13:47.442 / 20160526 09:14:05.838 / 00:00:18.396
00:00:18.395 KEYWORD ${resp} = rest_client . OpenBMC Post Request /org/openbmc/records/events/action/clear, data=${NIL}
Start / End / Elapsed: 	20160526 09:13:47.443 / 20160526 09:14:05.838 / 00:00:18.395
00:00:00.001 KEYWORD ${base_uri} = BuiltIn . Catenate SEPARATOR=, ${DBUS_PREFIX}, ${uri}
00:00:00.000 KEYWORD ${headers} = BuiltIn . Create Dictionary Content-Type=application/json
00:00:00.001 KEYWORD Collections . Set To Dictionary ${kwargs}, headers, ${headers}
00:00:00.002 KEYWORD rest_client . Log Request method=Post, base_uri=${base_uri}, args=&{kwargs}
00:00:02.777 KEYWORD rest_client . Initialize OpenBMC
00:00:15.613 KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Post Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

Send a POST request on the session object found using the
Start / End / Elapsed: 	20160526 09:13:50.225 / 20160526 09:14:05.838 / 00:00:15.613
09:13:55.231 	WARN 	Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",)': /org/openbmc/records/events/action/clear 	
09:13:55.231 	INFO 	Starting new HTTPS connection (2): 9.3.23.32 	
09:14:05.837 	INFO 	/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/.tox/barreleye/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
  InsecureRequestWarning)
/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/.tox/barreleye/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
  InsecureRequestWarning) 	
09:14:05.838 	FAIL 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /org/openbmc/records/events/action/clear (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",)) 	
===========================================================================

causten, 2016-05-26:
May 25 16:20:21 barreleye obmc-rest[976]: Exception in thread Thread-2:
May 25 16:20:21 barreleye obmc-rest[976]: Traceback (most recent call last):
May 25 16:20:21 barreleye obmc-rest[976]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
May 25 16:20:21 barreleye obmc-rest[976]: self.run()
May 25 16:20:21 barreleye obmc-rest[976]: File "/usr/lib/python2.7/site-packages/rocket/worker.py", line 179, in run
May 25 16:20:21 barreleye obmc-rest[976]: handled = self._handleError(*exc)
May 25 16:20:21 barreleye obmc-rest[976]: File "/usr/lib/python2.7/site-packages/rocket/worker.py", line 98, in _handleError
May 25 16:20:21 barreleye obmc-rest[976]: if 'timed out' in val.args[0]:
May 25 16:20:21 barreleye obmc-rest[976]: TypeError: argument of type 'int' is not iterable

williamspatrick, 2016-06-17:
Is this the same a #307 ?

williamspatrick, 2016-06-30:
There have been fixes for REST timeouts now.  Please retest.

closed 2016-06-30 by williamspatrick