openbmc/openbmc #307: Test Suite: Test Obmcrest: Inconsistency

sivassrr, 2016-05-26:
Same system and same version of image -  Test suite, Test Obmcrest gives different answers.

Success case: 

Full Name: 	Tests.Test Obmcrest
Documentation: 	

This suite will verifiy all OpenBMC rest interfaces Details of valid interfaces can be found here... https://github.com/openbmc/docs/blob/master/rest-api.md
Source: 	/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/tests/test_obmcrest.robot
Start / End / Elapsed: 	20160526 09:34:31.131 / 20160526 09:36:50.195 / 00:02:19.064
Status: 	50 critical test, 48 passed, 2 failed
50 test total, 48 passed, 2 failed

**Failed Test Case Details:**
Test Case: get names/

Message: 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",))

{ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

=========
Test Case: get names /org/

Send a GET request on the session object found using the
Start / End / Elapsed: 	20160526 09:35:47.919 / 20160526 09:35:59.156 / 00:00:11.237
09:35:52.925 	WARN 	Retrying (Retry(total=0, connect=None, read=None, redirect=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",)': /org/enumerate 	
09:35:52.926 	INFO 	Starting new HTTPS connection (3): 9.3.23.32 	
09:35:59.154 	INFO 	/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/.tox/barreleye/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
  InsecureRequestWarning)
/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/.tox/barreleye/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
  InsecureRequestWarning) 	
09:35:59.155 	FAIL 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /org/enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",))

===============================================================================

With the same hardware, Same image, Restarted the test with power on and waited for some time. 
Failed with all tests in this suite. 

This suite will verifiy all OpenBMC rest interfaces Details of valid interfaces can be found here... https://github.com/openbmc/docs/blob/master/rest-api.md
Source: 	/Users/sivas/OpenBMCAutomation/openbmcauto24May2016/openbmc-automation/tests/test_obmcrest.robot
Start / End / Elapsed: 	20160526 10:48:15.757 / 20160526 10:58:21.023 / 00:10:05.266
Status: 	50 critical test, 0 passed, 50 failed
50 test total, 0 passed, 50 failed
=================================

sivassrr, 2016-05-27:
Reran obmcrest and some improved number now. 

Full Name: 	Tests.Test Obmcrest
Documentation: 	

This suite will verifiy all OpenBMC rest interfaces Details of valid interfaces can be found here... https://github.com/openbmc/docs/blob/master/rest-api.md
Source: 	/Users/sivas/OpenBMCAutomation/openbmcauto26May2016/openbmc-automation/tests/test_obmcrest.robot
Start / End / Elapsed: 	20160526 21:35:12.422 / 20160526 21:38:35.016 / 00:03:22.594
Status: 	50 critical test, 48 passed, 2 failed
50 test total, 48 passed, 2 failed

sivassrr, 2016-06-03:
Ran couple of times today and yesterday 1 time with FW8.1 and tested on barreleye systems, Rest APIs worked. Not seeing bigger / basic failures.

sivassrr, 2016-06-03:
root@barreleye:~# cat /etc/os-release ; cat /etc/version
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.8.1-8-gbfcf3bf-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.8.1
20160602031800

Will do multiple regression using automated script and see the results then close it.

sivassrr, 2016-06-08:
Ran with Jun 6th Barreleye Driver. 
out of 10 iterations, only 3 Passed. Previously with Jun 3rd driver out of 5 iterations, 1st 3 iteration found some result.  After that all tests failed. From test suite, BMC not able to login.  But whereas if I login manually getting following error. Need to fix it.  (Even though entered the password correctly). 
 ssh root@9.3.23.24
Warning: Permanently added '9.3.23.24' (RSA) to the list of known hosts.
root@9.3.23.24's password: 
Permission denied, please try again.
root@9.3.23.24's password: 
Permission denied, please try again.
root@9.3.23.24's password: 
Permission denied (publickey,password).

williamspatrick, 2016-06-10:
@bradbishop - Do we have the root password correctly being set in the non-Barreleye machines?

sivassrr, 2016-06-16:
Previous iteration same test cases passed. And today I logged in one of the Palmetto system found no issue in root password. 
palmetto login: root
Password:
root@palmetto:~# cat /etc/version
20160606174128
root@palmetto:~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.8.1-11-gdef271f-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.8.1
root@palmetto:~# pwd
/home/root
root@palmetto:~#

Will try with the latest FW and kick off my long run test.

sivassrr, 2016-06-16:
Based on ST with Brad, He mentioned that he is working on moving to new http server some time next week, that will help seems.

closed 2016-06-30 by williamspatrick

sivassrr, 2016-07-04:
On Palmetto - All REST API Passed. But on Barreleye still 2 TCs related to "get names /" , "get names /org/" failing as follows:
=============================================
Full Name: 	Test Obmcrest & Test Eventlog & Test Syslog & Test Led & Test Generic Conf & Test Ssl & Test Fw Version.Test Obmcrest.get names /
Start / End / Elapsed: 	20160704 11:30:04.358 / 20160704 11:30:17.108 / 00:00:12.750
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",))
00:00:12.748 KEYWORD ${resp} = rest_client . OpenBMC Get Request /enumerate
Start / End / Elapsed: 	20160704 11:30:04.359 / 20160704 11:30:17.107 / 00:00:12.748
00:00:00.001 KEYWORD ${base_uri} = BuiltIn . Catenate SEPARATOR=, ${DBUS_PREFIX}, ${uri}
00:00:00.003 KEYWORD rest_client . Log Request method=Get, base_uri=${base_uri}, args=&{kwargs}
00:00:01.422 KEYWORD rest_client . Initialize OpenBMC
00:00:11.320 KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

Send a GET request on the session object found using the
Start / End / Elapsed: 	20160704 11:30:05.786 / 20160704 11:30:17.106 / 00:00:11.320
11:30:17.106 	FAIL 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",)) 	
00:00:19.571 TEST get names /org/
Full Name: 	Test Obmcrest & Test Eventlog & Test Syslog & Test Led & Test Generic Conf & Test Ssl & Test Fw Version.Test Obmcrest.get names /org/
Start / End / Elapsed: 	20160704 11:30:17.111 / 20160704 11:30:36.682 / 00:00:19.571
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /org/enumerate (Caused by ReadTimeoutError("HTTPSConnectionPool(host='9.3.23.32', port=443): Read timed out. (read timeout=5.0)",))
00:00:19.570 KEYWORD ${resp} = rest_client . OpenBMC Get Request /org/enumerate
Start / End / Elapsed: 	20160704 11:30:17.111 / 20160704 11:30:36.681 / 00:00:19.570
00:00:00.001 KEYWORD ${base_uri} = BuiltIn . Catenate SEPARATOR=, ${DBUS_PREFIX}, ${uri}
00:00:00.004 KEYWORD rest_client . Log Request method=Get, base_uri=${base_uri}, args=&{kwargs}
00:00:08.081 KEYWORD rest_client . Initialize OpenBMC
00:00:11.483 KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

Send a GET request on the session object found using the
Start / End / Elapsed: 	20160704 11:30:25.198 / 20160704 11:30:36.681 / 00:00:11.483
11:30:36.681 	FAIL 	ConnectionError: HTTPSConnectionPool(host='9.3.23.32', port=443): Max retries exceeded with url: /org/enume

=======================================
root@palmetto:~# cat /etc/os-release; cat /etc/version ;
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0 (krogoth)"
VERSION_ID="v1.0-48-g8d43aee-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (krogoth)"
BUILD_ID="v1.0"
20160630213138
root@palmetto:~#
======================================