openbmc/openbmc #309: Pflash hits miscompare  @ 0x01c87600 - Used both May27 and May28 images

sivassrr, 2016-05-30:
Barreleye BMC: 9.3.23.32:

Tried May27 and May28 image of barreleye and found hitting Micompare when we use Pflash. 
====================================================================
root@barreleye:/tmp# pflash -b -E -p ./flash-barreleye-20160527122028
About to erase chip !
Erasing... (may take a while !) done !
About to program "./flash-barreleye-20160527122028" at 0x00000000..0x02000000 !
Programming & Verifying...
[=============================================     ] 89% ETA:14s     LIBFLASH: Miscompare at 0x01c87600
Verification failed for chunk at 0x01c80000
root@barreleye:/tmp#

=================================================

root@barreleye:/tmp# pflash -b -E -p ./flash-barreleye-20160528115647
About to erase chip !
Erasing... (may take a while !) done !
About to program "./flash-barreleye-20160528115647" at 0x00000000..0x02000000 !
Programming & Verifying...
[=============================================     ] 89% ETA:14s     LIBFLASH: **Miscompare** at **0x01c88a00**
Verification failed for chunk at 0x01c80000

nkskjames, 2016-06-01:
You can only use pflash to flash BMC if the BMC is booted in ramfs mode.   Normally the BMC uses the flash as its filesystem, so flashing it while running would be bad.    Also, pflash is not really the prescribed way to flash because it only supports a full flash and doesn't preserve network settings.  So I'm closing this issue.    You should use to the initramfs update method via REST or manually.

closed 2016-06-01 by nkskjames

sivassrr, 2016-06-01:
Sure, In case if the same thing happens via Rest or Manuall code update, May I know what are all the data generally needed for debugging. Could you provide it.