openbmc/openbmc #310: PreTag 0.8 Testing failure for 'Tests.Test Time.Set Valid System Time'

gkeishin, 2016-05-30:
Set Valid System Time
Full Name: 	Tests.Test Time.Set Valid System Time
Documentation: 	

**GOOD PATH** This test case tries to set system time using IPMI and then tries to cross check if it is correctly set in BMC. Expectation is that BMC time should match with new time.
Start / End / Elapsed: 	20160530 10:23:04.427 / 20160530 10:23:06.844 / 00:00:02.417
Status: 	FAIL (critical)
Message: 	Open BMC time does not match with set time

**********************************

Start Time Test case :   	${stdout} = 05/30/2016 04:53:04

Set date   = 2016-02-29 09:10:00

But the Test set failed to update  and returns = 05/30/2016 04:53:06

Documentation: 	Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160530 10:23:04.429 / 20160530 10:23:05.664 / 00:00:01.235
10:23:04.429 	INFO 	Executing command '/tmp/ipmitool -I dbus sel time set "02/29/2016 09:10:00"'. 	
10:23:05.664 	INFO 	Command exited with return code 0. 	
10:23:05.664 	INFO 	${stdout} = 05/30/2016 04:53:04 	
10:23:05.664 	INFO 	${stderr} = 	
10:23:05.664 	INFO 	${output} = 0

This can be easily recreated manually ..

Assuming the ipmitool binary is already there in the system,

root@palmetto:~# hostname
palmetto
root@palmetto:~# date
Mon May 30 09:30:01 UTC 2016
root@palmetto:~# /tmp/ipmitool -I dbus sel time set "02/29/2016 06:10:00"     <-- Set time 
05/30/2016 09:30:08
root@palmetto:~# date
Mon May 30 09:30:11 UTC 2016     <--- No change 
root@palmetto:~#

gkeishin, 2016-05-30:
The same seems to have passed in Barreleye 
==============================================================================
Test Time :: This suite is for testing System time in Open BMC.               
==============================================================================
Set Valid System Time :: ***GOOD PATH***                              | PASS |
------------------------------------------------------------------------------
Test Time :: This suite is for testing System time in Open BMC.       | PASS |
1 critical test, 1 passed, 0 failed
1 test total, 1 passed, 0 failed
==============================================================================

This also could mean there could be issue with the RTC  chip on the Palmetto box this test is run.. 

Will try on another Palmetto system and confirm...

vishwabmc, 2016-05-30:
I am not too sure if we are supporting everything in Palmetto. If this works on Barreleye, need to take that as a baseline -or- upgrade the Palmetto to latest one.

gkeishin, 2016-05-30:
On Barrelleye ; BMC Version

{
  "data": {
    "fault": "False", 
    "fru_type": "BMC", 
    "is_fru": 0, 
    "manufacturer": "ASPEED", 
    "present": "False", 
    "version": "v0.8-1-gbdf51ec-dirty"
  }, 
  "message": "200 OK", 
  "status": "ok"
}

On the Parmetto

{
  "data": {
    "fault": "False", 
    "fru_type": "BMC", 
    "is_fru": 0, 
    "manufacturer": "ASPEED", 
    "present": "False", 
    "version": "v0.8-3-g03f51f5-dirty"
  }, 
  "message": "200 OK", 
  "status": "ok"
}

Looks the Palmetto is on the Latter version 8.3  than the Barrelleye 8.1..  Looks like its already upgraded..

gkeishin, 2016-06-13:
We can see this on latest [System - barreleye Build - 20160610155319 ] ..

Please look into it. Thanks

williamspatrick, 2016-06-30:
Need someone to confirm the ipmi time routines work (or not).

vishwabmc, 2016-07-07:
I ran some tests and the problem is with syncing the hwclock to system time.. 

root@barreleye:/usr/lib/host-ipmid# ./ipmitool  -I dbus sel time set "08/11/2016 09:10:00"
08/11/2016 09:10:00

root@barreleye:/usr/lib/host-ipmid# date
Thu Jul  7 07:31:06 UTC 2016

root@barreleye:/usr/lib/host-ipmid# hwclock -r
Thu Aug 11 09:10:09 2016  0.000000 seconds
---------------------------------------------------------------------

I tried syncing the HW time to system time but that fails to do so :
root@barreleye:/usr/lib/host-ipmid# hwclock --hctosys
root@barreleye:/usr/lib/host-ipmid# 

root@barreleye:/usr/lib/host-ipmid# date
Thu Jul  7 07:33:50 UTC 2016
root@barreleye:/usr/lib/host-ipmid# hwclock  -r
Thu Aug 11 09:13:32 2016  0.000000 seconds

==================================

Aug 11 09:10:00 barreleye dbus[960]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.80" (uid=0 pid=1362 comm="./ipmitool -I dbus sel time set 08/11/2016 09:10:0") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.78" (uid=0 pid=1350 comm="/usr/sbin/ipmid ")
Aug 11 09:10:00 barreleye dbus-daemon[960]: dbus[960]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.80" (uid=0 pid=1362 comm="./ipmitool -I dbus sel time set 08/11/2016 09:10:0") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.78" (uid=0 pid=1350 comm="/usr/sbin/ipmid ")

Those are what gets printed in journalctl.. I am looking more..

vishwabmc, 2016-07-08:
Finally it was due to systems-timesyncd that we were seeing the time getting reset. 

We have 2 options to solve this particular issue :

1/ Do not start systemd-timesyncd
2/. Reject this and handle it as part of  https://github.com/openbmc/openbmc/issues/98 since this can not be fixed without we agree on common design for using NTP / systemd-timesyncd.

mdmillerii, 2016-07-08:
3) query and stop service in test setup, restore at cleanup.

Note I might have wrong term but systemd has stop (temp) and disable (configure etc to prohibit) and this only requires the temporary stop.

williamspatrick, 2016-07-08:
It is a common configuration that the management network (BMC) cannot access the internet and therefore cannot use the default ntp servers.  In that case, the IPMI path is used because the ntp-client is ran on the Host-OS.  In our test environment, the BMC can access the external internet and therefore can get to the ntp servers.  Likely, if both sides can get to the real network, then the time settings from local ntp and ipmi-indirect are the same value.

I think the final solution should be delayed until #98 is at least designed.  We should then work with the test team on the appropriate way to test this.

In the meantime, should they even use the `date` command as part of the testcase?  There is no requirement that the BMC and Host timestamps are in sync.  The only requirement is that get-sel-time and set-sel-time are consistent and emulate a hardware RTC.  Should the testcase be changed to use get-sel-time instead of date?

vishwabmc, 2016-07-08:
the get-sel-time does no good either since it calls into time() call which is anyway overwritten by the NTP. 

So I suggest the test do these in this order :
1/. systemctl stop systemd-timesyncd
2/. Send that ipmi command to set-time
3/. Send that ipmi command to get-sel time

kc8apf, 2016-07-08:
I'm unclear if ipmi sel time get/set are intended to only modify the BMC's RTC or the BMC's kernel time.  The former can be done with hwclock --set --date which avoids touching the kernel clock.  If the kernel time is supposed to be changed as well, there needs to be some sort of mutual exclusion between impi sel time set and timesyncd.  Changing the test to temporarily disable timesyncd fixes the test without fixing the bug.

mdmillerii, 2016-07-08:
@kc8apf I understand your point, but we also have no setup of systemd-timesyncd beyond whatever the dhcp client provides via systemd-networkd integration in the systemd framework.  

I figure that turning it off for the test addresses the testing issue and if a deployment doesn't want ntp they can disable it in their build, but both its running and its configuration are things that should be addressed in #98.  

Disabling for the duration of the test provides similar coverage to changing the IP and MAC address and then changing it back to the prior values.

Currently there is one concept of time and it is both kernel and sel time.  There is a battery backed rtc chip on i2c for the hardware clock and the rtc in the ast2400 (reset on full chip reset and not battery powered) is disabled in the device tree.  As @williamspatrick stated this time is also the rtc used by the host.

kc8apf, 2016-07-08:
@mdmillerii The test failing is showing that the assumptions about SEL time being mapped onto BMC kernel time are either incorrect or broken when other time synchronization systems are enabled.  I'm arguing that, if timesyncd is enabled at all, the interactions between SEL time set and timesyncd need to be designed.  Leaving timesyncd enabled in the build and hoping that deployments won't allow NTP servers to be reachable via the BMC network is going to cause problems later.  Specifically, that won't work for Google's deployments.

williamspatrick, 2016-07-09:
The full design of the clock is something @vishwabmc will be working on this month. We had a proposal for a few different modes, one of which is an emulated split clock between the host and the Bmc. He will send the details to the mailing list for comment and discussion.

gkeishin, 2016-07-11:
This would be addressed using https://github.com/openbmc/openbmc/issues/98    as per the discussion thread here and input from @vishwabmc  as well... so closing this one...

closed 2016-07-11 by gkeishin