openbmc/openbmc #311: PreTag 0.8 Centaur0 not Present - Is this System Dependent Use case Test

gkeishin, 2016-05-31:
Failed on Palmetto System but  on Barreleye this test case passed..
 
 [ Please refer : https://github.com/mkumatag/openbmc-automation/issues/64 ]
 
 Seems like
 '/tmp/ipmitool -I dbus raw 0x04 0x30 255 0x00 0x00 0x00 0x00 0x40 0x00 0x00 0x20 0x00'
 
 Doesn't disable the property so it was still accessible as present..
 
Test Log for reference 
 -----------------------------------------------
Full Name: 	Tests.Test Sensors.Centaur0 not Present
Start / End / Elapsed: 	20160530 23:33:47.069 / 20160530 23:33:52.313 / 00:00:05.244
Status: 	FAIL (critical)
Message: 	True != False
00:00:00.000 KEYWORD ${uri} = BuiltIn . Set Variable /org/openbmc/inventory/system/chassis/motherboard/membuf0
Documentation: 	

Returns the given values which can then be assigned to a variables.
Start / End / Elapsed: 	20160530 23:33:47.072 / 20160530 23:33:47.072 / 00:00:00.000
23:33:47.072 	INFO 	${uri} = /org/openbmc/inventory/system/chassis/motherboard/membuf0 	
00:00:00.013 KEYWORD ${x} = Get Inventory Sensor Number ${uri}
Start / End / Elapsed: 	20160530 23:33:47.073 / 20160530 23:33:47.086 / 00:00:00.013
00:00:00.002 KEYWORD ${x} = model . Get Inventory Sensor ${OPENBMC_MODEL}, ${name}
23:33:47.086 	INFO 	${x} = 255 	
00:00:01.288 KEYWORD ipmi_client . Run IPMI Command 0x04 0x30 ${x} 0x00 0x00 0x00 0x00 0x40 0x00 0x00 0x20 0x00
Start / End / Elapsed: 	20160530 23:33:47.087 / 20160530 23:33:48.375 / 00:00:01.288
00:00:01.285 KEYWORD ${output}, ${stderr} = SSHLibrary . Execute Command /tmp/ipmitool -I dbus raw ${args}, return_stderr=True
Documentation: 	

Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160530 23:33:47.087 / 20160530 23:33:48.372 / 00:00:01.285
23:33:47.088 	INFO 	Executing command '/tmp/ipmitool -I dbus raw 0x04 0x30 255 0x00 0x00 0x00 0x00 0x40 0x00 0x00 0x20 0x00'. 	
23:33:48.371 	INFO 	Command exited with return code 0. 	
23:33:48.372 	INFO 	${output} = 	
23:33:48.372 	INFO 	${stderr} = 	
00:00:00.000 KEYWORD BuiltIn . Should Be Empty ${stderr}
00:00:00.001 KEYWORD BuiltIn . Set Test Variable ${OUTPUT}, "${output}"
00:00:03.928 KEYWORD Read the Attribute ${uri}, present
Start / End / Elapsed: 	20160530 23:33:48.376 / 20160530 23:33:52.304 / 00:00:03.928
00:00:03.925 KEYWORD ${output} = rest_client . Read Attribute ${uri}, ${parm}
Start / End / Elapsed: 	20160530 23:33:48.377 / 20160530 23:33:52.302 / 00:00:03.925
00:00:03.921 KEYWORD ${resp} = rest_client . OpenBMC Get Request ${uri}/attr/${attr}
Start / End / Elapsed: 	20160530 23:33:48.378 / 20160530 23:33:52.299 / 00:00:03.921
00:00:00.001 KEYWORD ${base_uri} = BuiltIn . Catenate SEPARATOR=, ${DBUS_PREFIX}, ${uri}
00:00:00.006 KEYWORD rest_client . Log Request method=Get, base_uri=${base_uri}, args=&{kwargs}
00:00:03.446 KEYWORD rest_client . Initialize OpenBMC
00:00:00.458 KEYWORD ${ret} = RequestsLibrary.RequestsKeywords . Get Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

Send a GET request on the session object found using the
Start / End / Elapsed: 	20160530 23:33:51.833 / 20160530 23:33:52.291 / 00:00:00.458
23:33:52.290 	INFO 	Get Request using : alias=openbmc, uri=/org/openbmc/inventory/system/chassis/motherboard/membuf0/attr/present, headers=None 	
23:33:52.291 	INFO 	${ret} = <Response [200]> 	
00:00:00.007 KEYWORD rest_client . Log Response ${ret}
Start / End / Elapsed: 	20160530 23:33:52.292 / 20160530 23:33:52.299 / 00:00:00.007
00:00:00.002 KEYWORD ${msg} = BuiltIn . Catenate SEPARATOR=, Response code:, ${resp.status_code}, , Content:, ${resp.content}
00:00:00.003 KEYWORD rest_client . Logging ${msg}, console=True
23:33:52.299 	INFO 	${resp} = <Response [200]> 	
00:00:00.001 KEYWORD ${content} = RequestsLibrary.RequestsKeywords . To Json ${resp.content}
Documentation: 	

Convert a string to a JSON object
Start / End / Elapsed: 	20160530 23:33:52.300 / 20160530 23:33:52.301 / 00:00:00.001
23:33:52.301 	INFO 	To JSON using : content={
  "data": "True", 
  "message": "200 OK", 
  "status": "ok"
} 	
23:33:52.301 	INFO 	To JSON using : pretty_print=False 	
23:33:52.301 	INFO 	${content} = {u'status': u'ok', u'message': u'200 OK', u'data': u'True'} 	
23:33:52.302 	INFO 	${output} = True

causten, 2016-06-01:
Fixed with https://github.com/mkumatag/openbmc-automation/pull/55
This can be closed

gkeishin, 2016-06-01:
Thanks.. Closing this ticket

closed 2016-06-01 by gkeishin