openbmc/openbmc #314: Always enable both watchdogs

mdmillerii, 2016-06-06:
For systems with dual flash chips, the aspeed SOC supports 2 watchdogs: The first is for normal boot timeout, and the second is configured with a longer timeout and an option to redirect chip select 0 to chip select 1 on the Static Memory Controller (Flash).

We should always expose both watchdogs to Linux.  Also need to sanitize the configuration.

shenki, 2016-06-06:
Currently the ast2500 uboot tree has watchdog 2 running. The ast2400 tree has watchdog 1 running.

I think we want to define our api between u-boot and Linux as watchdog 2 is always running. In systems where a second bmc flash exists the second watchdog can be used to fail over to a the second flash.

anoo1, 2017-05-31:
With the ubifs implementation, per Patrick this should not be needed.

shenki, 2017-06-01:
This is not related to ubifs. As discussed on the call today, we will need to enable the watchdogs.

geissonator, 2017-10-05:
Per discussion last night, this is currently enabled.  Will create follow on story to enable using it form userspace.  #2427 tracks enabling this from userspace.

closed 2017-10-05 by geissonator