openbmc/openbmc #315: Firestone / Garrison - Hold PCI resets until the processor takes over

anoo1, 2016-06-07:
Hold the PCI resets until hostboot sets the boot progress other than BASE_INITIALIZATION (0x14) which is the first progress code.

williamspatrick, 2016-06-17:
Need investigation if this is applicable to Witherspoon as well.  If not, we may drop the priority to medium.

mdmillerii, 2016-06-17:
Its not an issue for Witherspoon.

williamspatrick, 2016-06-17:
Do you know why it is a problem on these two boards but not others?  Is this a p8 flaw or something special about those two boards?  How are we sure Witherspoon is not affected?

mdmillerii, 2016-06-20:
The PCI specification requires the reset signal to be active (low) until the power and reference clock have been stable for some defined time (100ms from memory).

The POWER8 and POWER8NVL processors provide for a reset signal and reference clock forwarding, but can only drive the reset active, and will only drive these signals after its drivers are enabled by `hostboot` in `istep7`.  Experience has shown some adapters fail if reset is released without clocks applied and do not recover with further application of reset but only with a full power cycle.  This requires other logic in the system to hold the reset active from the time power is applied through the SBE initialization phase and into the `istep7` procedure, which is after the first message `BASE_INITIALIZATION` mentioned above has been issued (at least for secondary processor chips).  The reset needs to be released to processor control before the `payload` is started by `hostboot` for normal PCI device operation.

After this design requirement surfaced, some systems (including Palmetto) were able to make use of the delayed software control of the memory domain power enable that occurs in `istep14` as an additional reset condition, while other systems relied on software control by their BMC or equivalent.  It was later determined that for this hardware choice to be effective, the memory domain power enable needed to be cleared by the power cycle, as code did not have a place to turn it off during a (normal or abnormal) shutdown.   In the Palmetto design, the I2C expander is on switched system power and self clearing upon application of power, meeting the second requirement.  Neither of the Firestone or Garrison systems were able to meet all of these requirements and instead rely on the BMC software to control a GPIO to hold the reset during the described window.

The processor in Witherspoon includes a change to drive the PCI resets to the inactive state to allow the system to cover this window.

williamspatrick, 2016-06-20:
Thanks Milton for the in-depth analysis.  I'm going to lower the priority to medium with this in view.  Those systems are only meant as a development vehicle and most of us don't need to boot the system up far enough where PCI is a concern.

adamliyi, 2016-08-09:
Currently in OBMC, there is code in power_control_obj.c:
```
		//if changed, set property and emit signal
		if(gpio != control_power_get_pgood(control_power))
		{
			control_power_set_pgood(control_power,gpio);
			if(gpio==0)
			{
				control_power_emit_power_lost(control_power);
				control_emit_goto_system_state(control,"HOST_POWERED_OFF");
				rc = gpio_open(&pcie_reset);
				rc = gpio_write(&pcie_reset,0);
				gpio_close(&pcie_reset);

				rc = gpio_open(&usb_reset);
				rc = gpio_write(&usb_reset,0);
				gpio_close(&usb_reset);

			}
			else
			{
				control_power_emit_power_good(control_power);
				control_emit_goto_system_state(control,"HOST_POWERED_ON");
				rc = gpio_open(&pcie_reset);
				rc = gpio_write(&pcie_reset,1); 
				gpio_close(&pcie_reset);

				rc = gpio_open(&usb_reset);
				rc = gpio_write(&usb_reset,1);
				gpio_close(&usb_reset);
			}
		}
```
But when power on host, we need to set PCIE_RESET to active low, why in the code we write '1' to PCIE_RESET? Did I misunderstand it?
```
rc = gpio_write(&pcie_reset,1);
```

bradbishop, 2016-08-10:
Hi Adam

There are patches to power_control_obj.c in the garrison layer, so the code in skeleton isn’t what you end up with on a Garrison/Firestone system.

I have a feeling the patches probably don’t apply at the moment.

-brad

adamliyi, 2016-08-19:
Thanks @bradbishop. I've found the patch.
But I am blocked on "How can BMC know about the Hostboot booting stage (istep)". Andriana gave me suggestions, and @mdmillerii do you know that?

">But how can BMC know current Hostboot booting stage? Is there any way to communicate with Hostboot?
We know when hostboot starts up and starts requesting IPMI msgs but not sure how BMC could read their progress codes."

vishwabmc, 2016-08-19:
@adamliyi Are you looking for a programmatic way to know -or- want to watch the things ?. 

@shenki The obmc-console : is there a way it can also get the console output from the hostboot ?. When I used it, it was the host kernell that I logged in to. 

@williamspatrick Is HB leaving some footprints anywhere in pnor / processor scratch on where is it ? and if so, is there a way to get that with no FSI links here ?

mdmillerii, 2016-08-19:
@adamliyi have your process monitor (subscribe to) the IPMI traffic.  Look for the boot progress sensor message.  If the message is not BASE_INITIALIZATION then its time to release the reset.  You can then wait until  the next time to power on the system to apply the reset.

mdmillerii, 2016-08-19:
@adamliyi Alternatively, create a notification when the sensor is changed and subscribe to that.

adamliyi, 2016-08-27:
Thanks Milton and Vishwa. There is "BootProgress" sensor. Hostboot will send IPMI set sensor command to BMC and update the boot progress. As the BMC specs says:
```
1.12 Boot Progress
Keeping track of the boot progress of a server can help in debug and monitoring.
The boot progress will be tracked in two different, yet complementary methods.
The first is Hostboot and OPAL updating the System Firmware Progress and the
OS Boot sensors. The second is by emitting the detailed progress codes to the
SOL console during Hostboot and OPAL initialization of the system.
Hostboot Requirements
R3.2-3.15 Update SOL with detailed IBM progress codes Phase 1
R3.2-3.16 Update “System FW Progress” sensor with appropriate
mappings as boot progresses (14h = isteps 6-9, 01h = isteps 10-14,
03h = isteps 15-16, 13h = isteps 17-21) Phase 4
```

adamliyi, 2016-09-08:
Fixed committed for review: https://gerrit.openbmc-project.xyz/#/c/414/

adamliyi, 2016-09-08:
[0001-Add-power-control-support-for-the-Garrison-system.patch.txt](https://github.com/openbmc/openbmc/files/461690/0001-Add-power-control-support-for-the-Garrison-system.patch.txt)

amboar, 2016-09-09:
@adamliyi why are we not patching skeleton directly? @mdmillerii suggests this behaviour is specific to Firestone and Garrison - maybe we should add some abstractions to skeleton to allow for different behaviours.

Not that I want to continue expanding skeleton's capabilities...

adamliyi, 2016-09-11:
@amboar , we already have a patch in obmc for Garrison and Firestone: https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone/recipes-phosphor/skeleton/skeleton/0001-Add-power-control-support-for-the-Garrison-system.patch
My patch will add code to this patch.
And currently I cannot verify this patch on Firestone HW due to issue: https://github.com/openbmc/openbmc/issues/533.

amboar, 2016-09-12:
@adamliyi I understand, but I was asking about the bigger picture here. We own both openbmc/openbmc and openbmc/skeleton; why are we patching one from the other? My opinion is this approach makes the system as a whole harder to understand for not much (any?) gain.

That's a bit off-topic though, so let's not get bogged down.

madscientist159, 2016-10-14:
I think we're hitting this issue on our Firestone machine; certain PCIe cards directly attached to the CPUs (i.e. not via the PCIe bridge) are randomly working / not working after power down / power up (actually, they seem to fail over 80% of the time).

Will try the patch from @adamliyi shortly but I share the same concerns in general about us patching something we already have control over -- Palmetto should be ported to the new power_control_obj format and the patch should be directly applied to skeleton.

adamliyi, 2016-10-14:
@madscientist159 , please note the patch: "https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone/recipes-phosphor/skeleton/skeleton/0001-Add-power-control-support-for-the-Garrison-system.patch", has been removed in git master. I am working to updating my patch.

madscientist159, 2016-10-14:
@adamliyi I did see that shortly after commenting.  Do you know when you might have an updated patch?

adamliyi, 2016-10-16:
@madscientist159, I should be able to finish before 10/18. If you could try on older code base, using my previous patch - just to verify the patch can really fix your issue, that would be great. I have no Firestone at hand so I don't know how to tell the PCIe devices working or not.

adamliyi, 2016-10-17:
Hi @madscientist159 , I just updated the patch at: https://gerrit.openbmc-project.xyz/#/c/414/ and https://gerrit.openbmc-project.xyz/#/c/415/.  Please have a look. I tried it with github master head on Firestone, but host CPU still cannot boot. Looks the fix for https://github.com/openbmc/openbmc/issues/533 still not merged. Please let me know if you have any issue.

madscientist159, 2016-10-17:
@adamliyi Unfortunately a quick update to the latest OpenBMC version seems to have caused more problems than expected.  This is going to have to wait until I can get time to debug Firestone again, and this probably won't happen until some kind of resolution to https://github.com/open-power/hostboot/issues/53 has been found.

madscientist159, 2016-10-17:
@adamliyi That being said, your patch looks like it should do the right thing.  Further comments are on Gerrit.

adamliyi, 2016-10-26:
Put the fix as a patch to skeleton: https://gerrit.openbmc-project.xyz/#/c/891/
@williamspatrick Any suggestion to integrate system specific code into skeleton?

geissonator, 2016-11-09:
Not merged so moving to next sprint.

adamliyi, 2016-11-18:
@williamspatrick , @geissonator , the fixes for this issue:
https://gerrit.openbmc-project.xyz/#/c/891/
https://gerrit.openbmc-project.xyz/#/c/415/
Has been reviewed, could you merge and close this issue?

closed 2016-11-18 by williamspatrick