openbmc/openbmc #320: PreTag: [System - barreleye  Build - 20160610155319 ] Disable Syslog

gkeishin, 2016-06-13:
While Running the test code .. we found that our test looks for the REST Object /org/openbmc/LogManager/rsyslog/action/Disable     Which it doesn't find it.. 

Please let us know if this is expected or if the object path we are using in the test needs to be changed or updated..

Logs and some inputs for your reference

Disable Syslog
    [Documentation]     ***GOOD PATH***
    ...                 This testcase is to verify disabling syslog.\n

    ${resp} =    Disable Syslog Setting    ${EMPTY}
    Should Be Equal    ${resp}    ok          <---- Failed 
	
 ${ret} = RequestsLibrary.RequestsKeywords . Post Request openbmc, ${base_uri}, &{kwargs}
Documentation: 	

Send a POST request on the session object found using the
Start / End / Elapsed: 	20160613 12:34:04.695 / 20160613 12:34:04.778 / 00:00:00.083
12:34:04.777 	INFO 	Post Request using : alias=openbmc, uri=/org/openbmc/LogManager/rsyslog/action/Disable, data={"data": [""]},                     headers={u'Content-Type': u'application/json'}, files=None, allow_redirects=True 	
12:34:04.778 	INFO 	${ret} = <Response [404]>

root@barreleye:~# cat /etc/version 
20160610155319

}rango@ubuntu:~/TOOLS_OPENBMC/tools/obmc_dvt$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET https://9.3.23.32//org/openbmc/
{
  "data": [
    "/org/openbmc/control", 
    "/org/openbmc/managers", 
    "/org/openbmc/HostServices", 
    "/org/openbmc/settings", 
    "/org/openbmc/UserManager", 
    "/org/openbmc/buttons", 
    "/org/openbmc/records", 
    "/org/openbmc/inventory", 
    "/org/openbmc/watchdog", 
    "/org/openbmc/sensors", 
    "/org/openbmc/HostIpmi"
  ], 
  "message": "200 OK", 
  "status": "ok"

rahulmah, 2016-06-14:
@hramasub : Syslog related test are failing. Seem like syslog server code is still not in the build. Please confirm.

Response code:404, Content:{
  "data": {
    "description": "org.openbmc.objectmapper.Error.NotFound: path or object not found: /org/openbmc/LogManager/rsyslog"
  }, 
  "message": "404 Not Found", 
  "status": "error"

williamspatrick, 2016-06-17:
Issue moved to [openbmc/openbmc-test-automation #25](https://github.com/openbmc/openbmc-test-automation/issues/25) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-06-17 by williamspatrick