openbmc/openbmc #321: PreTag: [System - barreleye Build - 20160610155319 ]  Set Sensor Boot progress

gkeishin, 2016-06-13:
The test case is getting what is expecting   baseboard init but its gets other progress state..
**FW Progress, Starting OS** != **FW Progress, Baseboard Init**

Please let us know what is the correct State it should be in this use case... in case if this was expected.. Thanks

For your reference ::

Set Sensor Boot progress
    ${uri} =    Set Variable    /org/openbmc/sensors/host/BootProgress
    ${x} =      Get Sensor Number   ${uri}

    Run IPMI command  0x04 0x30 ${x} 0xa9 0x00 0x04 0x00 0x00 0x00 0x00 0x14 0x00
    Read the Attribute  ${uri}    value
    Response Should Be Equal    FW Progress, Baseboard Init
	
	
	Documentation: 	

Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160613 12:32:55.784 / 20160613 12:32:56.321 / 00:00:00.537
12:32:55.785 	INFO 	Executing command '/tmp/ipmitool -I dbus raw 0x04 0x30 5 0xa9 0x00 0x04 0x00 0x00 0x00 0x00 0x14 0x00'. 	
12:32:56.320 	INFO 	Command exited with return code 0. 	
12:32:56.320 	INFO 	${output} = 	
12:32:56.321 	INFO 	${stderr} = 

Start / End / Elapsed: 	20160613 12:33:08.090 / 20160613 12:33:08.091 / 00:00:00.001
12:33:08.091 	FAIL 	FW Progress, Starting OS != FW Progress, Baseboard Init

12:33:08.086 	INFO 	To JSON using : content={
  "data": "FW Progress, Starting OS", 
  "message": "200 OK", 
  "status": "ok"
} 	
12:33:08.086 	INFO 	To JSON using : pretty_print=False 	
12:33:08.086 	INFO 	${content} = {u'data': u'FW Progress, Starting OS', u'status': u'ok', u'message': u'200 OK'} 	
12:33:08.087 	INFO 	${output} = FW Progress, Starting OS

rahulmah, 2016-06-14:
@gkeishin : Test is passing with latest build [20160613175742]. We can close it

gkeishin, 2016-06-15:
Closing the behavioral error here. Will re-open if we hit again on this. .Thanks

closed 2016-06-15 by gkeishin