openbmc/openbmc #325: Concurrent pflash build can fail

amboar, 2016-06-14:
Saw the subsequent failure:

```
2 16:23:26 andrew@keelia:~/src/openbmc/openbmc/build$ bitbake obmc-phosphor-image
WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |#########################################################################################################################################| Time: 00:00:35
Parsing of 943 .bb files complete (0 cached, 943 parsed). 1382 targets, 246 skipped, 0 masked, 0 errors.
NOTE: Stamp /home/andrew/src/openbmc/openbmc/build/tmp/stamps/armv5e-openbmc-linux-gnueabi/pflash/5.2.1-r0 is not reachable, removing related manifests
NOTE: Stamp /home/andrew/src/openbmc/openbmc/build/tmp/stamps/palmetto-openbmc-linux-gnueabi/linux-obmc/4.4+gitAUTOINC+6717e8fec2-r0 is not reachable, removing related manifests
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "palmetto"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta
meta-yocto
meta-phosphor
meta-aspeed
meta-ast2400
meta-openpower
meta-ibm
meta-palmetto     = "HEAD:9f1d336d6708a63c7c59f8938bc0d4f365d4671a"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: oe_runmake failed
ERROR: Function failed: do_compile (log file is located at /home/andrew/src/openbmc/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.2-r0/temp/log.do_compile.5134)
ERROR: Logfile of failure stored in: /home/andrew/src/openbmc/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.2-r0/temp/log.do_compile.5134
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 CROSS_COMPILE=arm-openbmc-linux-gnueabi- PFLASH_VERSION=5.2.2 V=1 -C external/pflash all LINKAGE=dynamic
| make: Entering directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.2-r0/git/external/pflash'
| ln -sf ../../libflash ./libflash
| ln -sf ../../ccan ./ccan
| ln -sf ../common ./common
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -c pflash.c -o pflash.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -c progress.c -o progress.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I.  -c common/arch_flash_common.c -o common-arch_flash_common.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I.  -c common/arch_flash_arm.c -o common-arch_flash_arm.o
| make -C ../shared
| make[1]: Entering directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/pflash/5.2.2-r0/git/external/shared'
| ln -sf ../../libflash ./libflash
| ln -sf ../../ccan ./ccan
| ln -sf ../common ./common
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c libflash/file.c -o libflash-file.o
| common/arch_flash_arm.c:33:17: fatal error: ast.h: No such file or directory
|  #include "ast.h"
|                  ^
| compilation terminated.
| ../../external/common/rules.mk:71: recipe for target 'common-arch_flash_arm.o' failed
| make: *** [common-arch_flash_arm.o] Error 1
| make: *** Waiting for unfinished jobs....
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c libflash/libflash.c -o libflash-libflash.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c libflash/libffs.c -o libflash-libffs.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c libflash/ecc.c -o libflash-ecc.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c libflash/blocklevel.c -o libflash-blocklevel.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c common/arch_flash_common.c -o common-arch_flash_common.o
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c common/arch_flash_arm.c -o common-arch_flash_arm.o
| ln -sf ../../hw/ast-bmc/ast-sf-ctrl.c common/ast-sf-ctrl.c
| ln -sf ../../include/ast.h common/ast.h
| ln -sf arch_flash_arm_io.h common/io.h
| arm-openbmc-linux-gnueabi-gcc  -march=armv5e -marm  -mthumb-interwork --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/palmetto  -O2 -pipe -g -feliminate-unused-debug-types -O2 -Wall -I. -Werror -Wall -g2 -ggdb -I. -fPIC  -c common/ast-sf-ctrl.c -o common-ast-sf-ctrl.o
[... 42 lines cut ...]
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "palmetto"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta
meta-yocto
meta-phosphor
meta-aspeed
meta-ast2400
meta-openpower
meta-ibm
meta-palmetto     = "HEAD:9f1d336d6708a63c7c59f8938bc0d4f365d4671a"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: QA Issue: /usr/sbin/control@underscore@bmc.exe_skeleton contained in package skeleton requires libopenbmc_intf.so, but no providers found in its RDEPENDS [file-rdeps]
WARNING: QA Issue: /usr/lib/host-ipmid/libstrgfnhandler.so_host-ipmid-fru contained in package host-ipmid-fru requires libwritefrudata.so, but no providers found in its RDEPENDS [file-rdeps]
NOTE: Tasks Summary: Attempted 2037 tasks of which 1834 didn't need to be rerun and all succeeded.

Summary: There were 3 WARNING messages shown.
2 16:40:22 andrew@keelia:~/src/openbmc/openbmc/build$
```

bradbishop, 2016-06-21:
I've submitted a patch upstream for this.

closed 2016-06-21 by williamspatrick

williamspatrick, 2016-06-21:
@bradbishop - I see here you mentioned you submitted upstream.  I'm not seeing it on the skiboot mailing list.  Are you sure it went through?  Maybe it was sent from a non-subscribed email account and is awaiting moderation.

bradbishop, 2016-06-22:
I goofed my git-send-email sendemail settings.  https://lists.ozlabs.org/pipermail/skiboot/2016-June/003644.html