openbmc/openbmc #343: Power / Cooling - Power Limit / Capping

anoo1, 2016-06-14:


rfrandse, 2017-03-29:
work is done for phase 4 with #1213

bradbishop, 2017-05-24:
Open questions

Is a DCMI equivalent REST API sufficient or are additional levels of control / reporting needed? (Todd).
Is someone working on DCMI and if so how far down the stack are they going?

geissonator, 2017-05-30:
BMC provides the user power cap to HTMGT via inband sensor during boot and HTMGT provides to occ when going active.  If customer sets power cap while OCC active, then BMC sends new power cap directly to master occ.

geissonator, 2017-07-05:
Capability to set a power cap via sysfs already in place.
DCMI interface which retrieves power cap and enable status done in #948.
Reducing this from 8->3 points and closing out #936 and #947

closed 2017-11-03 by rfrandse