openbmc/openbmc #352: RAS / MFG - SEL

anoo1, 2016-06-14:
Parse eSEL into OpenBMC native structure

vishwabmc, 2016-09-06:
@anoo1 how are we consuming the parsed data ?. Do you have more info on this ?

geissonator, 2017-01-05:
Take existing code which takes hostboot eSEL as input and parses into our own structure and generates our new event dbus object.
Convert to story, make 5 pts, put under error log epic

anoo1, 2017-01-11:
Port existing functionality from phosphor-event to phosphor-logging and use sdbusplus bindings:
- [ ] Create new sdbusplus methods for IPMI to call to create and retrieve SEL information.
- [ ] Create associations using new association object(?)
- [ ] Create delete method using new object manager.
- [ ] Cache and retrieve logs using new property cacher.
- [ ] Port handling of SEL reservation id, store partial SELs, and create an event dbus object.
- [ ] Update IPMI to call new methods.

Export eSEL information:
- [ ] Gather requirements on what eSEL data and format would the user need access to.

geissonator, 2017-01-12:
Notes from sprint planning:
- We have plans to provide SEL via IPMI, we should do this, not REST, ipmi get sel #443 is related
so we should move hostboot SEL/eSEL to new phosphor-logging
- We provide REST api to extract phosphor-logging events which contain the eSEL, then we run parsing tools to get eSEL info.
- How do we want to distinguish between host and bmc errors?  Unique name for host errors.
- New object manager to delete errors
- Need separate story for associations
- Focus on transferring what we have over to new stuff and using sdbusplus

closed 2017-02-17 by geissonator