openbmc/openbmc #354: RAS / MFG - Manufacturing / Characterization

anoo1, 2016-06-14:
Need definition on requirements.

geissonator, 2017-01-05:
Sized big due to a lot of unknowns.  There will be things cronus wont be able to do that openbmc will need to do.  Reducing to 40 based on current data.

geissonator, 2017-03-29:
OCC interfaces tracked under #345 
Factory Reset under #865 
BMC Dump under #868 
Power/Cooling under #343 #346 and #347 

Open Questions (and only potential remaining work)
- SBEFIFO pass through work required by openbmc?
- Cronus access to i2c devices while system booted (hwmon has devices bound).

Reduce this epic to 8pts to account for the above open items.

geissonator, 2017-04-17:
No additional SBEFIFO work, mfg should not have to unbind any devices.  Reduce from 8 to 3.

closed 2017-11-13 by zahrens