openbmc/openbmc #356: Security - BMC Image Signing

anoo1, 2016-06-14:
Implement ECDSA with openssl, hopefully leverage Secureboot signing infrastructure

kc8apf, 2016-07-08:
I've been investigating u-boot's verified boot infrastructure.  I'm hoping to send out a first pass of commits to implement signed images by 7/15/2016

kc8apf, 2016-07-19:
https://github.com/kc8apf/openbmc/tree/kernel_fitimage is enough to build a FIT image that contains a kernel+dtb.  Talking with @shenki on IRC, there are a few options for how to package the image and they influence how much flash layout needs to be encoded in the dtb.  The possible flash layouts are (u-boot, FIT, r/o root, r/w), (u-boot, FIT, r/w), (u-boot SPL, FIT, r/o root, r/w), (u-boot SPL, FIT, r/w).  @shenki and I prefer (u-boot SPL, FIT, r/w) which means FIT needs to contain (u-boot, kernel, dtb, r/o root).  That needs a new image_type for building a FIT image.  I've got a rough start on that.  I'm not sure when I'll get back to hacking on it.

anoo1, 2017-11-15:
* Auto-generate private/public key pair (ssh key gen?) during bitbake like the REST server does for certificates, then users can change their preferred provider to override (1pt).
* During build, package the image in a way to be able to be verified by u-boot, may need to pack the rofs in the fitimage (1pt).

TODO:
* What are there mechanisms in u-boot to verify images? Talk with Rick and Milton.

closed 2018-04-11 by rfrandse