openbmc/openbmc #358: Security

anoo1, 2016-06-14:


geissonator, 2018-05-01:
Some meeting minutes from a community call on this:

- Signed images
- FIT images
- Running services as non-root
- SELinux - Where we at with that upload?
  - SELinux policy settings
- User id and passwords (PAM, LDAP)
  - Make sure external interfaces have appropriate roles, privileges, and permissions
  - Lock someone out after certain invalid amount of logins
- Secure coding guidelines
  - Buffer overruns and such
- Locking down filesystem
  - only /var and /etc (hopefully remove etc as writeable)
- Firewall
  - Packet firewall - i.e. only let 10.x network in
  - Firmware firewall - switch IPMI over to read-only for inband path
  - whitelist IPMI commands
  - DBUS - read-only properties
- Running Security Test Tools
  - Different groups on call run a variety of tools, will publish results as allowed
- Misc
  - Kernel and other open source package updates for security
  - Removal of debug and insecure interfaces prior to shipping
  - Python itself is a security concern so need to work on getting all python apps out
  - Also, bash not really the most secure either, work on getting rid of that too
- Two use cases
  - OpenBMC hacker that just wants to download and start working, need to be sure we still allow this
  - Second is shipping a system that guarantees the customer it only runs what it's supposed to

There's still some community discussion going on with the web server of choice, Nginx is being brought into a machine layer and a lot of security discussions happened within the review (https://gerrit.openbmc-project.xyz/#/c/6879/).

geissonator, 2018-05-01:
FYI @joseph-reynolds

ojayanth, 2018-05-31:
**Firmware image Security:** 
	-Image signing: Process of digitally signing Open BMC firmware to ensure that the software has not been altered since it was signed and also helps to validates the source of the image. 
       Workgroup scope: Implementation is available.
	

- Explore system use cases
- OpenBMC hacker that just wants to download and start working, need to be sure we still allow this 
- Second is shipping a system that guarantees the customer it only runs what it's supposed to 

**Authentication:** 
	Multiple authentication schemes are available .
- Session based , 
- Cookie based , 
- OAuth based authentication    

Workgroup scope: Define the applicability of the various authentications scheme for the use-cases such as user( REST,REDFISH,IPMI,HMC) to BMC  and BMC to BMC communication 

**Access Control** 
User Management
- Make sure external interfaces have appropriate roles, privileges, and permissions 
- Lock someone out after certain invalid amount of logins 
- Support PAM and LDAP authentication. 
SELinux. access control security policy settings
- For every current user or process ( and Files, network ports, and other hardware) , SELinux assigns a three string context consisting of a username, role, and domain (or type). 
- A typical policy consists of a mapping (labeling) file, a rule file, and an interface file, that define the domain transition. These three files must be compiled together with the SELinux tools to produce a single policy file. 
Workgroup scope: 
- User Management: Community is already driving this currently.
- SELinux: Define policy settings for openBMC services and direction for specifying policy settings for new BMC services. 

**Certificate management:**   
- Provide interface to override  the default shipped certificates . 
- Self signing the certificates during genesis boot. 

**Filesystem Security:** 
- Explore the need to lock-down /etc as RO 

**Network Security:** 
- Packet firewall - based on IP address and port number. 
- Firmware firewall - switch IPMI over to read-only for inband path 
- whitelist IPMI commands 
- DBUS - read-only properties. 

**Trust:** 
- Trust among peer to peer BMC's 
- Role of POC BMC to establish trust. 
- What are the services which required trust 
- Are all the BMC's running same set of services. 

**Secure Engineering Guidelines:** 
- Secure coding guidelines 
- OWASP(Open Web Application Security Project) compliance. 
- Running Security Test Tools. 
- Kernel and open source package security update. 
- Removal of debug and insecure interfaces prior to shipping. 
- Python itself is a security concern so need to work on getting all python apps out. 
- Secure alternate for bash shell. 
Workgroup scope: 
- Defining code review checklist for security coding. 
- Exploring OWASP compliance tool in CI. 
- Define Security updates process.

joseph-reynolds, 2018-06-01:
Thanks for the details!  I've added much of the information in your update to the following sections of the OpenBMC security documentation:
 - obmc-protection-profile-modules / Secure BMC firmware images
 - obmc-protection-profile-modules / Secure payload host firmware images
 - obmc-security-framework / FIA_UAU User authentication
 - obmc-security-framework / FIA_USB User-subject binding
 - bmc-security-framework / FPT - protection of the BMC
 - bmc-security-framework / FDP - user data protection
 - bmc-security-framework / FDP_UIT inter-BMC data integrity transfer protection
 - bmc-security-framework  / ALC_FLR flaw remediation
 - bmc-security-framework  / ALC_LCD lifecycle definition
I'll push those those updates to gerrit in a few hours: https://gerrit.openbmc-project.xyz/#/c/10443/

closed 2018-08-01 by rfrandse