openbmc/openbmc #361: UI - Web UI

anoo1, 2016-06-14:


geissonator, 2017-01-05:
- Web sockets - 5 pts
- Serving static pages - 3 pts
Leave epic at 13 for any unkowns

geissonator, 2017-06-29:
Meeting minutes from today:

GUI Meeting
All issues getting put under #361

Main login page info - #1857
  o What is “server id”? - Action for GUI team - is it asset id, serial number, host name?
  o System serial number
  o hard code path for key things like system serial number
  o system has to boot first before we get a valid serial numbe
  o Need some design discussion on this ( #1867 )

Chassis power consumption
  o Define in a README the path to the hwmon object (need issue)
  o General - need this for everything

#1864 - Concern with host0?
  o Need to change hard coded paths in one of 2 ways
    o dynamically determine it
    o document that the path is required

pcap has value and enable settings dbus objects
  o Andrew to talk with Iffy on this

Getting BMC time
  o Lei has code coming that will provide this

LED’s not working on witherspoon? 

Serial over LAN console - need story
  o hterm is what’s used by GUI

BMC network settings - refactor is coming from Ratan1

#1768  - Cors uploading via GUI

User management
  o Need to be able to at least change root password
  o sub-story to https://github.com/openbmc/openbmc/issues/446 ?

iftekharuli, 2017-06-29:
Here is what Michael captured during our meeting. Thought I will keep them at the same place. 

**Login:** 
GUI will need to be able to handle if the pre-login info is not present

Add paths for server info once logged in to the story. Help ensure it’s correct on first try when implemented in back end

**Action item** 
Figure out what Server ID is for login: Hostname? 

**Power consumption/cap question:** 
Power cap is user set, so we have that. Unsure where power consumption comes from. Comes from OCC (on chip controller)?. Andrew Geissler to add path to API for power consumption.

**BMC time**
 Will be an API to get current BMC time. Units will be seconds & date and will be converted. 

Ping Brad on how to show sensor threshold values for our GUI (Brad says it’s easy. file in a folder that can be altered)

Check if LED Indicator is working on Witherspoon and BarrelEye. 

**Remote console** 
Backend team needs to add websockets to create connection. Mention we’re using hterm in issue so back-end team can know what library we’re using.

**Network settings** 
switching over to new interface soon. Need path and more info to wire up. Network is not in image yet. Find out next week when (someone) comes back and new network stuff is in.

**Firmware** 
It is possible to remove boot priority 0 and have just 1,2,3. You may not have one set to 0. One activated is lowest priority and does not have to be zero. No indicator for current image. Have to query and current image is one with lowest priority number.

Ping Brad about upload firmware button and CORS issue. Headers are not added for upload route

**User account**
Backend team has an epic for user management that they have not worked on yet. They may need to jump on that sooner. Andrew G. is taking on adding password management for GUI.

closed 2017-11-16 by zahrens