openbmc/openbmc #366: Boot on AST2500 EVB

kc8apf, 2016-06-15:
An image built with TEMPLATECONF=meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf should successfully boot to a login shell on an AST2500 EVB.

kc8apf, 2016-06-15:
openbmc/u-boot#7 sets the u-boot default boot args so console goes to ttyS4 (debug UART)
openbmc/linux#93 adds the flash layout to aspeed-ast2500-evb.dts so the rootfs can be found

With those merged, an image built from openbmc/openbmc master should boot to a login shell.

williamspatrick, 2016-06-17:
This is a tracker issue so not assigning a size estimate.

legoater, 2016-06-17:
It does boot correctly but we need the n/w driver fixes regarding the ring buffer bits in the 4.6 kernel.

legoater, 2016-06-17:
Here is the branch I use https://github.com/legoater/linux/commits/openbmc-4.6. Network works fine.

legoater, 2016-06-20:
I just updated the g5 defconfig in my linux branch. Some more configs are required to boot an evb-ast2500 image. With these, you should be able to flash and successfully boot the board.

GHF, 2016-06-21:
@legoater if you build with bitbake, it overrides the defconfig with its own. I haven't tried this one verbatim but it should work. https://github.com/openbmc/openbmc/blob/master/meta-openbmc-bsp/meta-aspeed/meta-ast2500/recipes-kernel/linux/linux-obmc/defconfig

legoater, 2016-06-22:
@GHF yes. I tend to put everything in the linux tree but it is not the way to go. The defconfig is correct under meta-ast2500.

shenki, 2016-08-30:
The device tree changes are made with the dev-4.7 kernel. Please reopen if you still experience issues.

closed 2016-08-30 by shenki