openbmc/openbmc #370: Fan control - Air cool vs Water cool policies.

williamspatrick, 2016-06-15:
Need to ensure we support different fan policies for air cooled vs water cooled.  On Witherspoon this will be controlled via a jumper.  Other systems may want a static MRW setting or a backplane VPD setting.

msbarth, 2016-12-30:
- [ ] Thermal team define water cooled zone & floor values(tables)
- [ ] Add water cooled zone & floor values(tables)
- [ ] Interface/read jumper(GPIO?)
- [ ] Initialize/set zone & floor application data to water cooled values when jumper set

bjwyman, 2017-03-22:
- [ ] Move zone, table, definitions and values to a new story.
- [ ] Define new interface for CoolingType
- [ ] Code to read jumper GPIO for water cooled or not.
- [ ] Update CoolingType property to indicate if water cooled or not

bjwyman, 2017-03-22:
Created openbmc/openbmc#1371 for the cooling zone tables and values.

bjwyman, 2017-04-11:
Needs #1387

bjwyman, 2017-04-28:
Recipes:
https://gerrit.openbmc-project.xyz/#/c/3990/

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3991 Create recipe files for phosphor-cooling-type
Resolves #370 Fan control - Air cool vs Water cool policies.

closed 2017-06-08 by williamspatrick