openbmc/openbmc #374: General parsing framework for MRW-XML

williamspatrick, 2016-06-16:


williamspatrick, 2016-06-17:
Maybe look at what is available from open-power/hostboot and open-power/serverwiz .

agangidi53, 2016-06-17:
so is the thought behind this to generate https://github.com/openbmc/skeleton/blob/5213a991a5a3bd107636f1b2cc3bbca560843462/configs/Barreleye.py  from  https://github.com/open-power/barreleye-xml/blob/master/barreleye.xml ? or just reuse the XML ?

williamspatrick, 2016-06-17:
@agangidi53 - I plan to change how we specify the machine information in openbmc (Barreleye.py today).  Some of the content in the system.py files today Brad is in the process of moving over to systemd service files.  Some of the content in there is also duplicated in the kernel device tree, for instance for the i2c devices.  I'd like to use the barreleye-xml as the prime source and then generate content from it as needed in openbmc.  Eventually, I hope we do not generate a single big file like Barreleye.py that describes the whole machine but generate smaller portions that are tailored to the consuming application.

closed 2016-09-13 by williamspatrick