openbmc/openbmc #380: Convert packages to autotools.

williamspatrick, 2016-06-17:


williamspatrick, 2016-06-30:
Assign to Matt Barth.

msbarth, 2016-07-18:
Preview of autotool source created for the openbmc/phosphor-host-ipmid repo
[autotools_phosphor-host-ipmid.tar.bz2.zip](https://github.com/openbmc/openbmc/files/369940/autotools_phosphor-host-ipmid.tar.bz2.zip)

msbarth, 2016-07-18:
Outline of steps to generate autotool config files & scripts:

autoscan
Rename configure.scan to configure.ac
Update configure.ac
   Add "AM_INIT_AUTOMAKE"
   Modify AC_INIT with proper field values
aclocal
autoheader
Create required automake files
   NEWS (`touch NEWS`)
   README (`touch README`)
   AUTHORS (`touch AUTHORS`)
   ChangeLog (`touch ChangeLog`)
   COPYING (`touch COPYING`)
automake --add-missing
autoconf

./configure --host=arm-openbmc-linux-gnueabi
./config.status
make

shenki, 2016-07-20:
Please submit the changes for review at gerrit. We do not share code by uploading zip files to bug trackers.

williamspatrick, 2016-09-06:
Split the work completed off into #544.  Changing the estimate of this story to 1pt.

msbarth, 2016-10-04:
- [ ] Convert phosphor-event

- [ ] Convert phosphor-objmgr

- [ ] Convert phosphor-hwmon

- [ ] Convert ipmi-fru-parser

- [ ] Convert inarp

- [ ] Convert btbridge

closed 2016-11-11 by williamspatrick