openbmc/openbmc #381: Propose and implement unit test framework

williamspatrick, 2016-06-17:


williamspatrick, 2016-06-17:
Look at GoogleTest / GoogleMock.
Need to understand what support is already in place in yocto (and does it matter?).

williamspatrick, 2016-06-30:
Assign to Matt Barth.

kc8apf, 2016-07-26:
googletest is designed to be used as a submodule within each project rather than having a shared, installed copy.  They have CMake support that make it really easy to integrate if you are using CMake.  Otherwise, they ask you follow the instructions in https://github.com/google/googletest/blob/master/googletest/README.md to add it to your build system directly.

geissonator, 2016-10-05:
### Sprint Planning Notes

- [ ] git submodule decision/implementation

williamspatrick, 2016-10-18:
Work completed.

closed 2016-10-18 by williamspatrick

shenki, 2016-10-18:
What did we end up going with?

Did we implement a project as an example?

williamspatrick, 2016-10-18:
Yes.  The host-ipmid package.