openbmc/openbmc #382: Add CI tests for all packages.

williamspatrick, 2016-06-17:


williamspatrick, 2016-06-17:
This is to ensure we have Jenkins running.  Requires #380 to give us a consistent test case invocation and #381 to give us the unit test framework.

williamspatrick, 2016-06-30:
Assign to Matt Barth.

williamspatrick, 2016-09-09:
The outcome of this should be a jenkins job for every C++ / autotools repository (hopefully templated for all the repositories) that compiles and runs the gtests.  #561 was opened to convert existing test cases to gtest.

geissonator, 2016-10-05:
- [ ] Create script which can run in docker container to do simple dependency item
- [ ] Script can handle complex dependencies accross repos
- [ ] Create template jenkins jobs that allows new repos to easily be added
- [ ] Automatically run jenkins job against gerrit changes

msbarth, 2016-10-31:
- [ ] Create script which can run in docker container to do simple dependency item

- [ ] Script can handle complex dependencies accross repos

- [ ] Create template jenkins jobs that allows new repos to easily be added

- [ ] Automatically run jenkins job against gerrit changes

closed 2017-01-18 by williamspatrick