openbmc/openbmc #388: Investigate btrfs

williamspatrick, 2016-06-17:


williamspatrick, 2016-06-17:
Need to understand the implications for mtd vs mtdblock.  btrfs is likely not mtd-aware so we need to understand if there are any bad performance or RAS side-effects of using mtdblock.

williamspatrick, 2016-06-17:
See #223 for additional discussion.

mdmillerii, 2016-06-22:
Some concerns or blockers with btrfs on mtdblock:

1.  No wear leveling.  It's not clear that the wandering logs will ever displace long term static data.  Since it hasn't been targeted at raw flash I doubt it has been a consideration.

2.  btrfs super blocks are at 64k, 64M, 256G, and 2T byte offsets.  That means a 32MB or even a 64MB partition only has one superblock copy that can be lost while writing.

3. No current lzma or xz compression.  (xz is  lzma with per block uncompressed bypass).  Only gzip and lzo are upstream (lzo is faster but less space efficient compression vs gzip).

4. The only logical block size supported is 4kB.  Not necessarily a problem for 4k subsector erase.

5.  Are there other documented users of btrfs on small volumes?  How much overhead is there to make it fast on larger current mainstream media?

I think 2 is unacceptable for our current systems and haven't heard of relief for future systems.

ubi adds volume ids and atomic block replacement but steals 2 64byte structures per erase block.  Ubi block emulation is read only because the IO is now over 2 blocks.  Jffs2 on glubi (mtd on ubi) gets a 4k-128 byte logical erase block (or 64k-128 with full sectors).

I don't think either 4k on ubi or not 4k on btrfs is a realistic out of tree patch.  An externally mapped ftl through dm could be but I don't know of any such projects, and it's not clear the fs is suitable for small volumes (32 to 64 MB total space).

williamspatrick, 2016-09-01:
We can also investigate "appended" or overlays of squashfs filesystems.  It doesn't look like squashfs performs block deduplication while it is building an image so that might be a requirement in order to get the delta-image smaller.

Also, we should investigate changing our squashfs build parameters to increase the block size (128k default -> 1MB?) and add the xz arm filter (-Xbcj arm).  These two cause our current image to go from 14MB -> 12MB of size.

anoo1, 2017-05-31:
Will not implement since we're going with ubifs.

closed 2017-05-31 by anoo1