openbmc/openbmc #398: Clear one time boot overrides fails

vishwabmc, 2016-06-21:
root@palmetto:~# systemctl status clear-once -l
● clear-once.service - Clear one time boot overrides
I noticed this every-time on a Palmetto boot..

  Loaded: loaded (/lib/systemd/system/clear-once.service; static; vendor preset: enabled)
  Active: failed (Result: exit-code) since Mon 2016-06-20 16:15:57 UTC; 1h 21min ago
 Process: 908 ExecStart=/sbin/fw_setenv openbmconce (code=exited, status=1/FAILURE)
Main PID: 908 (code=exited, status=1/FAILURE)

Jun 20 16:15:54 palmetto systemd[1]: Starting Clear one time boot overrides...
Jun 20 16:15:55 palmetto fw_setenv[908]: Cannot access MTD device /run/fw_env: No such file or directory
Jun 20 16:15:55 palmetto fw_setenv[908]: Error: environment not initialized
Jun 20 16:15:57 palmetto systemd[1]: clear-once.service: Main process exited, code=exited, status=1/FAILURE
Jun 20 16:15:57 palmetto systemd[1]: Failed to start Clear one time boot overrides.
Jun 20 16:15:57 palmetto systemd[1]: clear-once.service: Unit entered failed state.

williamspatrick, 2016-06-21:
@mdmillerii - Can you take a look at this?  I suspect an fstab entry is missing.

mdmillerii, 2016-06-21:
@vishwabmc Can you confirm that you booted a cpio.lzma.uboot initramfs image?

If you booted from a normal flash path send private email with the machine details.

vishwabmc, 2016-06-21:
@mdmillerii - I used cpio.lzma.u-boot intramfs image... Yes

mdmillerii, 2016-06-21:
Ok so the issue is that the "fstab", `/etc/fw_env.cfg`, part of the `u-boot-fw-utils` package, is pointing to a not existing file and offset because the first half of the workaround for openbmc/linux#94 is implemented in the `obmc-phosphor-initfs` package `init` script that is skipped when using that image.   

There is already a check that `/etc/fw_env.cfg` is not empty but it can otherwise be invalid and I don't plan to fix that part.  (I expect it to be missing if the bootloader does not support u-boot).

1. I have a alternate image as part of my #297 pull request.  Although it _currently_ requires additional options to not use the `rwfs` mtd partition, it can be built to not do that and run from RAM with teh addition of one variable in a config file (or via the bootloader when choosing the image).

2. I can add a unit to replicate the logic to create the link in a `systemd` unit.

geissonator, 2018-07-18:
@mdmillerii - this fixed in master now?

mdmillerii, 2018-07-19:
I'd like someone to look at a boot log on palmento, but I think its good.

closed 2019-04-26 by stale[bot]