openbmc/openbmc #399: Add raspberry pi as a platform

shenki, 2016-06-21:
Enable the raspberry pi as a simple, cheap, easy to obtain development platform for OpenBMC.

We should use the [meta-raspberrypi layer](http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi) to obtain the board support bits.

https://lists.ozlabs.org/pipermail/openbmc/2016-June/003731.html

Related to #384.

williamspatrick, 2016-06-30:
Assign to @adamliyi

adamliyi, 2016-07-19:
Submitted a initial patch for review and test: https://lists.ozlabs.org/pipermail/openbmc/2016-July/004266.html
With this patch, the obmc-phosphor-image can be built for raspberrypi. Since I don't have access to Raspberrypi HW, I did not submit a formal patch including the meta-raspberrypi layer to openbmc Gerrit repo.

I can submit a formal patch once it is tested on HW.

williamspatrick, 2016-08-17:
Imported the layer into our repo:

remote:   https://gerrit.openbmc-project.xyz/225 Squashed 'import-layers/meta-raspberrypi/' content from commit 2745399

The significant part of work for this story isn't to just import the layer, but to make a raspberry-pi target for obmc-meta-phosphor.

williamspatrick, 2016-08-23:
@adamliyi - Can you rework your patch to utilize the import-layers/meta-raspberrypi layer I pulled in?  Then please push to Gerrit.

kachini, 2016-09-13:
folks, what is the status of this issue? i've got raspberry pi HW and can test if needed.

adamliyi, 2016-09-14:
@kachini , I just committed a patch to Gerrit for review:
https://gerrit.openbmc-project.xyz/#/c/508/

I have no HW so not tested. With this patch, I can build obmc images:

```
deploy/images/raspberrypi$ ls -l image-*
lrwxrwxrwx 1 adam adam 32 Sep 14 12:06 image-bmc -> flash-raspberrypi-20160914023347
lrwxrwxrwx 1 adam adam 57 Sep 14 12:06 image-initramfs -> core-image-minimal-initramfs-raspberrypi.cpio.lzma.u-boot
lrwxrwxrwx 1 adam adam  7 Sep 14 12:06 image-kernel -> cuImage
lrwxrwxrwx 1 adam adam 43 Sep 14 12:06 image-rofs -> obmc-phosphor-image-raspberrypi.squashfs-xz
lrwxrwxrwx 1 adam adam 10 Sep 14 12:06 image-rwfs -> rwfs.jffs2
lrwxrwxrwx 1 adam adam 10 Sep 14 12:06 image-u-boot -> u-boot.bin
```
Steps to build:

```
1. Put the configure files in import-layers/meta-raspberrypi/conf/
2. rm -fr build/conf
3. export TEMPLATECONF=import-layers/meta-raspberrypi/conf
4. . oe-init-build-env
5. bitbake obmc-phosphor-image
```
But I am thinking the image-* file is targeted for booting from NOR flash. If we are to boot from a SD card in Raspberrypi, we may need to change the image generation variables. Suggestions is welcome.

Arjun-Padmanabhan, 2016-09-20:
HI,

I executed bitbake with your changes.
Followed below mentioned steps:
1. Put the configure files in import-layers/meta-raspberrypi/conf/
2. rm -fr build/conf
3. export TEMPLATECONF=import-layers/meta-raspberrypi/conf
4. . oe-init-build-env
5. bitbake obmc-phosphor-image

Bitbake completed without issue #587.

Attaching files found in images directory.
file:///root/Desktop/openbmc/openbmc/build/tmp/deploy/images/raspberrypi3/files.txt
Also attaching local.conf file i used.
file:///root/Desktop/openbmc/openbmc/build/conf/local.conf

Can you let me know which file i need to flash to sd card for booting obtained image from raspberrypi?

williamspatrick, 2016-09-21:
@Arjun-Padmanabhan - I don't think @adamliyi actually has a RP3 to try out.  He was working on the commits to make an image build.  In theory the 'obmc-phosphor-image-raspberrypi3.rpi-sdimg' is what you would place on an SD card, but we have not attempted it yet to know if we set the config files up correctly for that.  @shenki was planning on trying it out as he had free time.  It is really a "spare time" activity for us right now and not a primary use case.

Arjun-Padmanabhan, 2016-09-28:
Hi,

I copied obmc-phosphor-image-raspberrypi3.rpi-sdimg and tried booting from sd card in RP3. 
RP3 is not booting up with the image in sd card

closed 2016-10-04 by williamspatrick

mungayree, 2018-04-23:
is it possible to provide some build steps to get latest yocto release + rpi ? I would like to try this over rpi for some bmc solution.

amboar, 2018-04-24:
@mungayree have you tried the steps provided by @Arjun-Padmanabhan in https://github.com/openbmc/openbmc/issues/399#issuecomment-248244943 (this is a genuine question, I haven't and I don't know if they work, but it's worth a shot)?

mine260309, 2018-04-24:
The "unknown" (and interesting) part is that RPi boots from a FAT partition by its GPU, but OpenBMC assumes u-boot, kernel and rofs partitions.

To make RPi work, I think we need to make a image that contains:
1. A FAT partition containing RPi's kernels and device tree blobs;
2. The root fs partition containing OpenBMC's rofs