openbmc/openbmc #400: 602031800: Barreleye build found hang in multiple system when trying to execute set sensor boot count

sivassrr, 2016-06-21:
root@barreleye:~# cat /etc/version
20160602031800
root@barreleye:~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v0.8.1-8-gbfcf3bf-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v0.8.1
root@barreleye:~#

Observed the below hang in both 9.3.23.30 and 9.3.23.32 Bareelye systems 
(BarreleyePVT &BarD-04 ) system which has the latest barreleye image 602031800 image. 
==============================================================
Tests.Test Sensors.Execute Set Sensor boot count
Start / End / Elapsed: 	20160621 15:31:51.861 / 20160621 19:03:25.139 / 03:31:33.278
Status: 	FAIL (critical)
Message: 	Execution terminated by signal
00:00:00.000 KEYWORD ${uri} = BuiltIn . Set Variable /org/openbmc/sensors/host/BootCount
00:00:01.335 KEYWORD ${x} = Get Sensor Number ${uri}
Start / End / Elapsed: 	20160621 15:31:51.862 / 20160621 15:31:53.197 / 00:00:01.335
00:00:01.333 KEYWORD ${x} = model . Get Sensor ${OPENBMC_MODEL}, ${name}
Start / End / Elapsed: 	20160621 15:31:51.863 / 20160621 15:31:53.196 / 00:00:01.333
15:31:53.196 	INFO 	${x} = 9 	
15:31:53.197 	INFO 	${x} = 9 	
03:31:31.942 KEYWORD ipmi_client . Run IPMI Command 0x04 0x30 ${x} 0x01 0x00 0x35 0x00 0x00 0x00 0x00 0x00 0x00
Start / End / Elapsed: 	20160621 15:31:53.197 / 20160621 19:03:25.139 / 03:31:31.942
03:31:31.939 KEYWORD ${output}, ${stderr} = SSHLibrary . Execute Command /tmp/ipmitool -I dbus raw ${args}, return_stderr=True
Documentation: 	

Executes `command` on the remote machine and returns its outputs.
Start / End / Elapsed: 	20160621 15:31:53.199 / 20160621 19:03:25.138 / 03:31:31.939
15:31:53.199 	INFO 	Executing command '/tmp/ipmitool -I dbus raw 0x04 0x30 9 0x01 0x00 0x35 0x00 0x00 0x00 0x00 0x00 0x00'.

=================================================

Previously this TC used to pass.

williamspatrick, 2016-06-21:
What does "602031800" mean?  I see this is still running v0.8.1 code:
```
VERSION_ID=v0.8.1-8-gbfcf3bf-dirty
```

Any reason you are not running v1.0?

sivassrr, 2016-06-22:
Patrick,
 
That was Jun 20th  image.
 
root@barreleye:~# cat /etc/version20160602031800    <================= Which was the latest image when I picked from https://openpower.xyz/job/openbmc-build/119/distro=ubuntu,target=barreleye/changes#detailroot@barreleye:~# cat /etc/os-releaseID=openbmc-phosphorNAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)VERSION=0.1.0 (jethro)VERSION_ID=v0.8.1-8-gbfcf3bf-dirtyPRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)BUILD_ID=v0.8.1root@barreleye:~#
 
 
Seeing yesterday image now - https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=barreleye/lastSuccessfulBuild/artifact/images/barreleye/  (barreleye-20160621200609.all.tar)
Sivas R. S
E-mail: sivas.srr@in.ibm.com
E2E FSP Test Architecture, OpenBMC Test Lead, Lab Management
9902233500
08046678273

williamspatrick, 2016-06-22:
@sivassrr - Those "build numbers" are not really useful to us.  They are just timestamps that bitbake creates when it runs on Jenkins, so they are effectively random numbers.  The VERSION_ID is what we need to point to a commit number.

v0.8.1-8-gbfcf3bf-dirty - means "v0.8.1" tag plus 8 commits, the last of which is "bfcf3bf", and the repository is "dirty" meaning the tool did some source level modification.

We should do some investigation as to why Jenkins is modifying the source.

That commit you are testing is now 20 days old.  So we need to make sure you are up on a more recent level.

williamspatrick, 2016-06-22:
```
git rev-list v1.0 ^bfcf3bf | wc
     49
```

This means there are 49 commits between what you are testing and v1.0.

sivassrr, 2016-06-27:
124 Build level able to run succcessfully 10 iterations with only 35 failures in each iteration. And same across the iteration. Consistency achieved. Today testing 125 build level, Will check and close it accordingly.

sivassrr, 2016-06-27:
curl command also fails

closed 2016-06-30 by williamspatrick