openbmc/openbmc #407: qemu: obmc-console.service fails to start

williamspatrick, 2016-06-22:
Journal:
```
Jun 22 19:06:04 qemux86-64 systemd[1]: obmc-console.service: Main process exited, code=exited, status=1/FAILURE
Jun 22 19:06:04 qemux86-64 systemd[1]: obmc-console.service: Unit entered failed state.
Jun 22 19:06:04 qemux86-64 systemd[1]: obmc-console.service: Failed with result 'exit-code'.
Jun 22 19:06:05 qemux86-64 systemd[1]: obmc-console.service: Service hold-off time over, scheduling restart.
Jun 22 19:06:05 qemux86-64 systemd[1]: Stopped OpenBMC console daemon.
Jun 22 19:06:05 qemux86-64 systemd[1]: obmc-console.service: Start request repeated too quickly.
Jun 22 19:06:05 qemux86-64 systemd[1]: Failed to start OpenBMC console daemon.
Jun 22 19:06:05 qemux86-64 systemd[1]: obmc-console.service: Unit entered failed state.
Jun 22 19:06:05 qemux86-64 systemd[1]: obmc-console.service: Failed with result 'start-limit'.
```

Running by hand:
```
root@qemux86-64:~# obmc-console-server 
obmc-console-server: Can't find parent device for ttyS5: No such file or directory
obmc-console-server: Can't access attribute sirq on device ttyS5: No such file or directory
obmc-console-server: Can't access attribute lpc_address on device ttyS5: No such file or directory
obmc-console-server: Can't open tty /dev/ttyS5: No such file or directory
```

williamspatrick, 2016-06-22:
This was running revision ed883a5d560d5c034925160458b3b5395a583d32

mdmillerii, 2016-06-30:
This is trying to talk to the vuart hardware.  I don't recall seeing that hardware being emulated yet.

vishwabmc, 2016-07-01:
* Jul 01 07:34:47 qemuarm obmc-console-server[444]: obmc-console-server: Can't open tty /dev/ttyS5: No such file or directory * 

Jul 01 07:34:47 qemuarm systemd[1]: obmc-console.service: Main process exited, code=exited, status=1/FAILURE
Jul 01 07:34:47 qemuarm systemd[1]: obmc-console.service: Unit entered failed state.
Jul 01 07:34:47 qemuarm systemd[1]: obmc-console.service: Failed with result 'exit-code'.
Jul 01 07:34:47 qemuarm systemd[1]: obmc-console.service: Service hold-off time over, scheduling restart.
Jul 01 07:34:47 qemuarm systemd[1]: Stopped OpenBMC console daemon.
Jul 01 07:34:47 qemuarm systemd[1]: obmc-console.service: Start request repeated too quickly.
Jul 01 07:34:47 qemuarm systemd[1]: Failed to start OpenBMC console daemon.

==============

Missing /dev/ttyS5 is why we have this failure.. Is it emulated at all ?. Do we see a need to emulate it ?. Are we trying to make QEMU a complete replacement to HW ?

williamspatrick, 2016-07-01:
We are not expecting QEMU to be a "complete replacement to harware" but it is a very useful development environment so I would like to see it as complete as it can be without modifying QEMU.

/dev/ttyS5 is the UART for the host.  If the device isn't present we shouldn't be launching the console-server, or the console server should have error handling to deal with the specified device being missing.  We might need to update systemd/udev to not error on this in QEMU when the device is not present.

vishwabmc, 2016-07-08:
I am trying to put a [Path] section and PathExists=/dev/ttyS5 for obmc-console.service file.
However, the socket based activation should have taken care of this anyhow.

mdmillerii, 2016-07-08:
Try to depend on the generated device instance like the getty service files.

williamspatrick, 2016-07-08:
Socket-based activation does not solve this.  If someone ssh's to the correct socket then systemd will attempt to start the service, which will fail at that time.  All it does is defer the identification of the problem.

williamspatrick, 2016-07-08:
@mdmillerii - Thanks for that pointer.

https://github.com/systemd/systemd/blob/09541e49ebd17b41482e447dd8194942f39788c0/units/serial-getty%40.service.m4

Here is a link to the original serial-getty@.service file.  We can probably change the host console program to a similar @.service file instead and have a machine variable that determines the host TTY.

kc8apf, 2016-07-12:
TTYs should create .device units that services can depend on.  See https://lists.fedoraproject.org/pipermail/devel/2012-January/160917.html.

vishwabmc, 2016-07-12:
@kc8apf Correct.  For right now, I am trying the path based activation and yes I am looking into the udev events based ones too.

vishwabmc, 2016-07-12:
The changes are almost done.. .Need some supporting change from systemd now that can do templatised instantiation that can be specified in the BB file.

vishwabmc, 2016-07-15:
@shenki has some concerns about the proposed change and passing the port via CLI and the #openbmc IRC has the discussion.. @shenki , @williamspatrick  and I need to have a discussion.

Meanwhile, if we agree to pass it by the CLI then I need a change that I proposed to @bradbishop where in I can do something like this below and SYSTEMD understands that.

HOST_TTY = "ttyS5"
SYSTEMD_SERVICE_${PN} = (BPN}@{HOST_TTY}.service.

At present, systemd / yocto build does not like this since it looks for obmc-console@tty5.service which wont be present..

shenki, 2016-07-19:
>
> @shenki has some concerns about the proposed change and passing the port via CLI and the #openbmc IRC has the discussion.. @shenki , @williamspatrick and I need to have a discussion.

I still haven't heard an argument as to why it needs to be on the
command line. The configuration file serves us fine at this point.

You can then make your proposed changes to yocto to generate the
configuration file that is installed to /etc.

williamspatrick, 2016-07-19:
In doing the investigation for this issue, it was discovered that systemd has a well-established mechanism for starting hardware-based processes and those are "template" units.  As mentioned above, we would write a 'obmc-console@.service' file and then start 'obmc-console@ttyS5' and the part after the @ becomes a parameter to the unit that can be passed along.  This is only "useful" if we have somewhere to pass it to.

One aspect of the template unit, which is appropriate to this issue, is that the template parameter can be used as a dependency, so we can ensure the service obmc-console@%s will not start unless /dev/%s is present.

There is no _need_ to make this change right now, but I still feel it is the appropriate change.
    1. It is, frankly, a pretty trivial feature to the obmc-console application.
    2. We can use systemd template unit files as intended without hard-coding the dev-entry.
    3. It gets us very close to being able to handle multiple host consoles for a blade-chassis style system.
    4. It allows us to easily map a [virtual] physical UART in QEMU to the obmc-console application for test purposes.

I realize that there is kernel dev-tree entries currently for the VUART, so this is not a complete solution.  @spinler is working on moving some of the dev-tree entries to generated from the MRW and/or udev based.  I expect the rest of the VUART work to be done in those features.

I see generating the /etc file for obmc-console and modifying obmc-console to accept an additional parameter as roughly the same amount of work.  I tend to favor the obmc-console C code from a maintenance perspective than more one-off yocto scripting.  Being able to override all foo.conf entries via command-line options is a fairly standard practice and also a feature applicable outside of our Yocto tree.

shenki, 2016-07-19:
> In doing the investigation for this issue, it was discovered that systemd has a well-established mechanism for starting hardware-based processes and those are "template" units. As mentioned above, [...]

Okay, that's a good enough justification to add it. Thanks for the
explanation Patrick.

Vishwa, please mention this in your commit message when justifying the
change. I look forward to reviewing your patch.

Cheers,

Joel

vishwabmc, 2016-07-20:
https://gerrit.openbmc-project.xyz/#/c/50/1

vishwabmc, 2016-08-02:
few updates : 

HOST_TTY = "ttyS5"

SYSTEMD_SERVICE_{PN} = "obmc-console@{HOST_TTY}.service will result in systemd creating a soft link in /etc/systemd/system/multi-user.target.wants/obmc-console@ttyS5.service to /lib/systemd/system/obmc-console@.service and it gets auto launched .. so its good.

However, we have failures in templated obmc-console-ssh@.service waiting on templated  obmc-console@.service.. 

So even with Requires=obmc-console@ttyS5.service and After=obmc-console@ttyS5.service inside obmc-console-ssh.socket results in obmc-console-ssh.socket starting even though the obmc-console@ttyS5 is not even running / fails to start.

I had a discussion on this in #systemd and they mentioned that what I have in there should have done the trick.

So, I am going to open a issue against systemd.

vishwabmc, 2016-08-02:
Opened : https://github.com/systemd/systemd/issues/3862

williamspatrick, 2016-08-16:
Waiting for changes from @bradbishop to support template systemd files.

williamspatrick, 2016-08-23:
@bradbishop - Is everything Vishwa needs up in Gerrit now?  Can you point him to it?

closed 2016-10-06 by williamspatrick