openbmc/openbmc #408: qemux86-64: obmc-console-ssh.socket failed to start

williamspatrick, 2016-06-22:
Running revision ed883a5d560d5c034925160458b3b5395a583d32

Journal 
```
# systemctl status obmc-console-ssh.socket -l
● obmc-console-ssh.socket - OpenBMC console ssh server socket
   Loaded: loaded (/lib/systemd/system/obmc-console-ssh.socket; enabled; vendor preset: enabled)
   Active: failed (Result: resources) since Wed 2016-06-22 19:06:03 UTC; 11min ago
   Listen: [::]:2200 (Stream)
 Accepted: 0; Connected: 0

Jun 22 19:06:02 qemux86-64 systemd[1]: Listening on OpenBMC console ssh server socket.
Jun 22 19:06:03 qemux86-64 systemd[1]: Closed OpenBMC console ssh server socket.
Jun 22 19:06:03 qemux86-64 systemd[1]: Stopping OpenBMC console ssh server socket.
Jun 22 19:06:03 qemux86-64 systemd[1]: obmc-console-ssh.socket: Failed to listen on sockets: Address already in use
Jun 22 19:06:03 qemux86-64 systemd[1]: Failed to listen on OpenBMC console ssh server socket.
Jun 22 19:06:03 qemux86-64 systemd[1]: obmc-console-ssh.socket: Unit entered failed state.
```

sivassrr, 2016-06-23:
root@barreleye:~# systemctl status obmc-console-ssh.socket -l
● obmc-console-ssh.socket - OpenBMC console ssh server socket
   Loaded: loaded (/lib/systemd/system/obmc-console-ssh.socket; enabled; vendor preset: enabled)
   Active: active (listening) since Thu 2016-06-23 06:15:54 UTC; 47min ago
   Listen: 0.0.0.0:2200 (Stream)
 Accepted: 0; Connected: 0

Jun 23 06:15:54 barreleye systemd[1]: Listening on OpenBMC console ssh server socket.
root@barreleye:~# cat /etc/version
20160622205331
root@barreleye:~# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0-4-gd8cd888-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0
root@barreleye:~#    (But all other test failing in this system 9.3.23.30)

vishwabmc, 2016-07-01:
root@qemuarm:/usr/sbin# cat /etc/os-release 
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0-13-gd6abfb9-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0
root@qemuarm:/usr/sbin# 

root@qemuarm:/usr/sbin# systemctl status obmc-console-ssh.socket
● obmc-console-ssh.socket - OpenBMC console ssh server socket
   Loaded: loaded (/lib/systemd/system/obmc-console-ssh.socket; enabled; vendor preset: enabled)
   Active: active (listening) since Fri 2016-07-01 07:34:47 UTC; 2h 53min ago
   Listen: [::]:2200 (Stream)
 Accepted: 0; Connected: 0

Jul 01 07:34:47 qemuarm systemd[1]: Stopping OpenBMC console ssh server socket.
Jul 01 07:34:47 qemuarm systemd[1]: Listening on OpenBMC console ssh server socket.

=== 
I am seeing its active on my QEMU on that levek.

williamspatrick, 2016-07-01:
The fact that your output has these implies that it failed at least once though:
Jul 01 07:34:47 qemuarm systemd[1]: Stopping OpenBMC console ssh server socket.
Jul 01 07:34:47 qemuarm systemd[1]: Listening on OpenBMC console ssh server socket.

vishwabmc, 2016-07-13:
I ran a Qemux86 today and it ran without any failures.

Looking at what was posted before while opening this feature, the key thing is : 
" obmc-console-ssh.socket: Failed to listen on sockets: Address already in use"

So it was the matter of 2200 already being used by some other.. ( but who ).. 

We may want to configure the TCP/IP setting to have 2200 fixed -or- change the range for ephemeral ports so that we can be safe. Although port in the range 2200 is not truly ephimeral but they are not fixed too.

vishwabmc, 2016-07-13:
root@qemux86:~# systemctl status obmc-console-ssh.socket -l
● obmc-console-ssh.socket - OpenBMC console ssh server socket
   Loaded: loaded (/lib/systemd/system/obmc-console-ssh.socket; enabled; vendor preset: enabled)
   Active: active (listening) since Wed 2016-07-13 12:48:29 UTC; 14min ago
   Listen: [::]:2200 (Stream)
 Accepted: 0; Connected: 0

Jul 13 12:48:29 qemux86 systemd[1]: Listening on OpenBMC console ssh server socket.
root@qemux86:~#

williamspatrick, 2016-07-13:
Your changes for #407 might be fixing this.  I suspect that the error I got was because of two things happening.

1. First instance exiting because unix socket not available to connect to obmc-console server.  (Need systemd unit file update to fix this).
2. Second instance, started automatically by systemd on first instance exit, [re]started too quick and tried to claim the port before the kernel cleaned up the old one.  We may need to rethink our server restart interval to deal with this.

vishwabmc, 2016-07-13:
For #1 : We may need to reserve the port 2200 in /etc/services ?.  We do not have any arbitrary process that may try to use 2200 thinking its ephemeral ?.. But we don't have any such process anyway. 

For #2 : I have seen this failure. So I worked around it by putting a RestartSec=3s.  So this needs a general discussion.

vishwabmc, 2016-07-13:
So in your case, looks like some other process already using 2200 and hence this .socket file is unable to LISTEN for incoming ssh sessions on that socket.

mdmillerii, 2016-07-13:
`/etc/services` will not reserve the port.  It only gives names for a port.

There is a `sysctl` file that can reserve specific ports, along with a second that controls the range of ports to assign to local ports.  See  `ip_local_reserved_ports`  and `ip_local_port_range` in [Documentation/networking/ip-sysctl.txt](http://github.com/openbmc/linux/tree/v4.6/Documentation/networking/ip-sysctl.txt#L791-828).

However, I suspect the issue is more likely related to getting `SO_REUSEADDR` and/or `SO_REUSEPORT` properly set on the socket either in the service or socket file.

The raw output in

    cat /proc/net/tcp

or formatted

    netstat -a

will show open and in-progress-closing sockets.

vishwabmc, 2016-07-14:
@mdmillerii I already checked the /proc/sys/net/ipv4/ip_local_port_range and that was not anyway near to 2200. So is why I was thinking about any possible alternatives. 

ip_local_reserved_ports did not have anything populated.

I will investigate on the usage of SO_REUSE

vishwabmc, 2016-07-15:
SO_REUSEPORT does not come to use here since the issue is about our very first attempt to LISTEN on 2200 getting to failure since someone else is already using it.

This looks like a one off and I have not been able to recreate this. Best I think we can do here is to reserve the port 2200 and close this issue.. Thought ?. @williamspatrick ?

vishwabmc, 2016-07-15:
We discussed about this in the scrum and the idea is to put this one on hold and then wait on changes for #407 and then try to recreate. If this does not get recreated, then we can close this one.

williamspatrick, 2016-08-16:
Waiting for #407 to be integrated so we can retest.

vishwabmc, 2016-11-09:
Is it still an issue ?

geissonator, 2016-11-23:
@williamspatrick Still an issue now that #407 is integrated?

geissonator, 2016-11-28:
Patrick, please reopen if still an issue.

closed 2016-11-28 by geissonator