openbmc/openbmc #409: qemux86-64: host-ipmi-hw using 100% of CPU

williamspatrick, 2016-06-22:
Running revision ed883a5d560d5c034925160458b3b5395a583d32

```
  254     1 root     R    97880  41%  99% python /usr/sbin/host-ipmi-hw
```

I'm not sure this program is even suppose to be here anymore.  Maybe the bt-host device is not created and this is poorly handling that.

williamspatrick, 2016-06-30:
This should be running btbridged maybe instead?  Or maybe no host-ipmi at all.

vishwabmc, 2016-07-01:
I looked at /usr/sbin/host-ipmi-hw and this is an emulator for BT in that it does implement the signal ( ReceivedMessage ) and method ( SendMessage ) that BT driver has implemented

ConsoleReader will watch for inputs from STDIN and on seeing data, it will fire the ReceivedMessage and IPMI daemon would catch and process and call SendMessage.

That said, 2 things : 

1/. Jeremy had ipmi_debug.py which did exactly this and I am looking for that now..
2/. the /bin/ps does not give the output that is mentioned in this issue. So which other PS was used ?

With how its written currently, I don't see why it would hog. I will try to look @ /proc contents

vishwabmc, 2016-07-01:
The 'top' gives :
  262     1 root     R    14572  12%  98% python /usr/sbin/host-ipmi-hw

killing it and restarting gives 12%.. 

To answer my #2 above, I got the ipmi_debug.py and what I saw was that ipmi_debug.py has been renamed as host-ipmi-hw.

I am thinking that the gobject.mainloop() is not effectively doing the stuff.. Will look more.

vishwabmc, 2016-07-07:
I propose to get rid of this since we have ipmitool now that does the same job and can be run on demand. I will create a PR to remove host-ipmi-hw from the image.

williamspatrick, 2016-07-07:
@bradbishop, @anoo1 - Any opinion on @vishwabmc's proposal?  I don't recall what this `host-ipmi-hw` is there for.

We don't have the ipmitool installed directly on the BMC, nor do I expect we will in the long term either.  It could be put into a devtools filesystem.  I suspect everything that this `host-ipmi-hw` program is doing can be replicated directly with `busctl` now.

vishwabmc, 2016-07-08:
@williamspatrick: I have explained the need for host-ipmi-hw in QEMU in the previous update. 

host-ipmi-hw is the BT driver emulator on QEMU so that we can try sending the IPMI commands. host-ipmi-hw has a console reader function that will watch for data from 'stdin' and then will fire a signal called "ReceivedMessage" ( the same that BT driver fires on a new IPMI packet ) and thus results in IPMID catching it and processing.

Once IPMID processes the Netfn/Cmd, it will call back into host-ipmi-hw using the sendMessage ( on the actual hardware, sendMessage is what is provided by BT driver ) and that marks the current IPMI netfn/cmd transaction as 'complete'.

Our custom 'ipmitool' does the exact same thing so is why I am proposing ipmi-host-hw to be removed. We are using ipmitool on the real HW anyway for all unit testing so I don't see we can live without that.

using the busctl. this would not be possible since we would need someone that can host a sendMessage API and ReceivedMessage signal on the HostIPMI dbus that IPMID can call utllize.

On the real HW, its the BT driver that is hosting those APIs and in this case, its host-ipmi-hw that is hosting those service and the same is hosted by ipmitool with a -I dbus channel.

williamspatrick, 2016-07-08:
I am ok with removing `host-ipmi-hw` but I still don't intend to include the custom `ipmitool`.  It is being used for testing as a way to short cut the testing so they don't need to boot up to the host-os, but that doesn't mean we should include it in the flash itself.  We will eventually have a devtools image that can include this.  Thank you for the explaination on why busctl is insufficient.

vishwabmc, 2016-07-08:
We agreed that we remove host-ipmi-hw. I will be opening a PR to get this removed.

vishwabmc, 2016-07-15:
@bradbishop Since the plan is to retain the virtual provider, I may not be able to remove the host-ipmi-hw-example.bb since sample.inc will now *not* have anything that can provide host-ipi-hw and hence BB fails since host-ipmid wants host-ipmih-hw providers. 

For right now, I have retained the BB file and then inside the ".service" file, I just have [Description] and that is helping build the image and also it does not start anything since ExecStart would not be there.

I will have to see if there are better ways of doing it

bradbishop, 2016-07-15:
This seems to have been fixed with 

https://github.com/openbmc/openbmc/commit/6f286af2d6e137498c87933445dbe512bb21bc00

vishwabmc, 2016-07-15:
So the change mentioned above deleted "inherit systemd" from the host-ipmi-hw-example and that is resulting in SYSTEMD not packaging the .service file and that solves this problem.

To make it a complete fix, I will remove SYSTEMD_SERVICE_{PN} and the host-ipmi-hw-example.service and we can close this issue.

vishwabmc, 2016-07-20:
https://gerrit.openbmc-project.xyz/#/c/41

closed 2016-07-27 by williamspatrick

bradbishop, 2016-08-21:
Hi! 

There is something important I need to  tell  you, that you'd  really like to hear, so please read it here <http://kislutoje.soundgecko.co/e4vud>

Kind regards, bradleyb

bradbishop, 2016-08-21:
Hey, 

I met my old friend yesterday, he is now so famous and talked by  a lot  of people. You can find more about him here  <http://vydogicy.elizabethstrianese.com/e4fih>

Speak to you later, bradleyb

bradbishop, 2016-08-21:
Dear, 

I came across that interesting stuff and I guess it's  something really crazy, just take a  look <http://treslephopy.gttechsecurity.com/e4htdcry>

Best wishes, bradleyb