openbmc/openbmc #416: Update u-boot config for Firestone.

williamspatrick, 2016-06-27:
See openbmc/u-boot#9 .

williamspatrick, 2016-06-30:
Need to get an updated u-boot config from @shenki and then update the recipes to utilize it.

williamspatrick, 2016-06-30:
Assign to Matt Barth

adamliyi, 2016-08-22:
I think this issue has been fixed by Joel's commit:
https://github.com/openbmc/u-boot/commit/f9b2a49f57c153c7358f9769234c0cefdf4de8d4 and https://github.com/openbmc/openbmc/commit/a244dcdcd45c0b55023095fb6cc3e5dcc0e090b6

Firestone is using u-boot configure without NS-CI, in  meta-openbmc-machines/meta-openpower/meta-ibm/meta-firestone/conf/machine/firestone.conf:
```
UBOOT_MACHINE_firestone = "ast_g4_phy_config"
```
This configure does not use NS-CI.
@shenki, do you think so?

shenki, 2016-08-29:
Good investigation adam. Yes, I think this bug can be closed once you verify that a current OpenBMC master image works on Firestone and the networking comes up.

adamliyi, 2016-09-14:
Verified using V1.99.0 obmc u-boot image on Firestone. The network work well. The u-boot configure file is "ast_g4_phy_config". So I think this issue has been fixed. Bellow are details:

1. network of u-boot in V1.99.0 obmc works well:
```
U-Boot 2016.07 (Sep 09 2016 - 15:45:16 +0800)

DRAM:  496 MiB
WARNING: Caches not enabled
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   MAC0 : RGMII
MAC1 : RMII/NCSI
FTGMAC100#0, FTGMAC100#1
Error: FTGMAC100#1 address not set.
Hit any key to stop autoboot:  0 
ast# print
baudrate=38400
bootargs=console=ttyS4,38400n8 root=/dev/ram rw
bootcmd=bootm 20080000 20300000
bootdelay=3
bootfile=all.bin
eeprom=y
ethact=FTGMAC100#0
ethaddr=70:E2:84:14:02:2C
gatewayip=9.3.183.1
ipaddr=9.3.183.183
netmask=255.255.255.0
serverip=9.3.164.219
stderr=serial
stdin=serial
stdout=serial
verify=n

Environment size: 343/131068 bytes
ast# ping 9.3.164.219
FTGMAC100#0: Starting autonegotiation...
FTGMAC100#0: Autonegotiation complete
FTGMAC100#0: link up, 100Mbps full-duplex
Using FTGMAC100#0 device
host 9.3.164.219 is alive
ast# tftp 0x40000000 cuImage_firestone
FTGMAC100#0: link up, 100Mbps full-duplex
Using FTGMAC100#0 device
TFTP from server 9.3.164.219; our IP address is 9.3.183.183; sending through gateway 9.3.183.1
Filename 'cuImage_firestone'.
Load address: 0x40000000
Loading: #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ############################################
	 1.2 MiB/s
done
Bytes transferred = 1555716 (17bd04 hex)
ast# 
```
2. u-boot is build with "ast_g4_phy_config":
```
NOTE: make -j 4 CROSS_COMPILE=arm-openbmc-linux-gnueabi- CC=arm-openbmc-linux-gnueabi-gcc  --sysroot=/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/firestone V=1 HOSTCC=gcc  -isystem/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/usr/lib -L/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/adam/workdir/openbmc_firestone/build/tmp/sysroots/x86_64-linux/lib -Wl,-O1 ast_g4_phy_config
```

adamliyi, 2016-09-14:
@williamspatrick , please close this issue.

closed 2016-09-14 by williamspatrick