openbmc/openbmc #417: Downloading unnecessary kernel

shenki, 2016-06-28:
I'm building post-yocto 2.1 for witherspoon and we still pull down a yocto kernel:

```
0: linux-yocto-4.4.3+gitAUTOINC+bcc6509084_ec9167bcb0-r0 do_fetch (pid 28923)
```

Can we override this to use the same tarball as the boot kernel?

shenki, 2016-06-28:
The version is defined in `yocto-poky/meta/recipes-kernel/linux/linux-yocto_4.4.bb`.

We have `meta-phosphor/common/recipes-kernel/linux/linux-obmc_4.6.bb` includes linux-yocto. We should be overriding the versions here, but we're not.

The download is extremley slow. It appears to be fetching  `http://downloads.yoctoproject.org/mirror/sources/git2_git.yoctoproject.org.linux-yocto-4.4.git.tar.gz`.

I wonder if the download is for building libc. The linux-libc-headers.inc file contains a strongly worded message discouraging anyone from using their own kernel tree to build libc. I disagree with their reasoning.

gtmills, 2018-05-23:
@shenki Is this still a problem? If not, can you close?

closed 2018-05-24 by shenki