openbmc/openbmc #418: Build fails to find libpcre

shenki, 2016-06-28:
Building 2d1d6b72143ce6d8bf4a5bbbb95eb28994876a54 on Ubuntu:

```
| ++ mapping --with-pcre=yes to --with-pcre="external:internal:none"
| ++ searching location: external
| -- extending CPPFLAGS: -I/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include
| -- extending LDFLAGS: -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib
| -- extending LIBS: -lpcre
| checking pcre.h usability... yes
| checking pcre.h presence... yes
| checking for pcre.h... yes
| checking for pcre_compile in -lpcre... no
| checking whether to build with PCRE library... no
| ++ executing failure action
| configure: error: unable to find usable PCRE library
| NOTE: The following config.log files may provide further information.
| NOTE: /home/joel/dev/obmc/openbmc/build/tmp/work/x86_64-linux/rpm-native/5.4.16-r0/build/config.log
| ERROR: configure failed
| ERROR: Function failed: do_configure (log file is located at /home/joel/dev/obmc/openbmc/build/tmp/work/x86_64-linux/rpm-native/5.4.16-r0/temp/log.do_configure.11100)
ERROR: Task 96 (virtual:native:/home/joel/dev/obmc/openbmc/meta/recipes-devtools/rpm/rpm_5.4.16.bb, do_configure) failed with exit code '1'
```

looking through config.log I find that it's configuring the rpm package with `--with-pcre`. It finds `pcre.h` and then performs this test:

```
configure:80099: checking for pcre_compile in -lpcre
configure:80124: gcc  -o conftest -isystem/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -DRPM_VENDOR_WINDRIVER -DRPM_VENDOR_POKY -DRPM_VENDOR_OE -g -D_GNU_SOURCE -D_REENTRANT -isystem/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -DRPM_OS_LINUX=040400 -I/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -I/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -I/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -DLTM_DESC -DUSE_LTM -I/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/include -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/lib -Wl,-O1 -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib  -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib  -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib  -L/home/joel/dev/obmc/openbmc/build/tmp/sysroots/x86_64-linux/usr/lib  conftest.c -lpcre  -ltomcrypt -lcrypto -lbz2 -lz -lpopt -lpthread  -lelf -lpopt -lz -lssl -lcrypto -ltommath -lpcre >&5
/usr/bin/ld: cannot find -ltommath
collect2: error: ld returned 1 exit status
```

So it's actually failing to link against `libtommath`?

shenki, 2016-06-28:
Upstream has a patch:

http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/recipes-devtools/rpm?id=71979b2ff7a0f78892435558e27bf0779367de55

williamspatrick, 2016-06-29:
Fixed with #424 .  Thanks Joel for the backport of a fix from Yocto-2.2 master.

closed 2016-06-29 by williamspatrick